slic3r/Slic3r #3660: More option in the object settings dialog

Tinchus2009, 2017-01-09:
### Version
latest dev version

### Behavior
First, I want to say that the modifier creation option recently added IS AWESOME.
This is a feature request. MAny times you have different object in the same plate for printing. In many cases I have noticed that support is being created but actually is not really necesary and that object could perfectly be printed useing bridges. The actual problem is that if you activate support generation, you will have it even if it si unnecesary, and that creates extra printing time. I know that of course I could print those objects alone. But would be nice if we have the hability to have the option to deactivate/activate support generation as another one of the option available in the object setting dialog.
Other settings that would be usefull to have in that dialog or in the modifier options dialog:
- Activate/deactivate/speed cooling (so we can control the cooling of specific areas, something really important with overhanhs for example)

_Is this a new feature request?_

Yes

I have ataches an image showing an example. Some of those object DO need support, some of them dont, I have marked the unnecesary supports. Some people may say that this is actually a problem of unecesary support generation, I really dont know, because that could also depend on the machine calibration and its ability to do bridges. So my thinking is just have to posibility of activate/deactivate support generation on an object basis.
[image]

lordofhyphens, 2017-01-09:
That is already a thing. Select the part in the Object Settings dialog and you can turn support generation on/off for that part only.

I don't remember if cooling is controllable via modifier meshes or the object settings.

Tinchus2009, 2017-01-09:
No, I cant. At least no in the latest dev version I builted 1 week ago. That option is not present in the obect settings dialog, neither in the modifier options. Cooling is not controllable neither by modifiers nor in the object settings

lordofhyphens, 2017-01-09:
[image]

Needs to be the Object Settings, not the Part. Sorry for the confusion.

lordofhyphens, 2017-01-09:
One workaround you could do is define multiple extruders for your printer with the different variants of filament profiles. Each extruder has an assigned "filament" profile.

You can assign the extruder with the filament profile that has the cooling method you want to modifier meshes.

I'm not disputing your request for override settings for cooling via modifier meshes, just mentioning an approach to get you going.

Tinchus2009, 2017-01-09:
Ok... there is a little bug I think in the UI. Look at the image I atached in my previous message. Those are all the options I have available, as you can see, all support options are not there. That list appears when I have the object's name selected it.
But if I click on the word "OBJECT" a lot of options became available. Is this suposed to be like this? Atached is another screenshot
[image]

lordofhyphens, 2017-01-09:
Yeah, I think that may be by design, but I don't know for sure. Individual parts in an object are being treated the same w/r/t available modifier meshes.

The "object" is a slightly higher level of abstraction. I think that it should be safe to make parts have same overrides as the overall object, but I haven't looked at it closely enough.

Tinchus2009, 2017-01-09:
Ok, you workaround suggestion would be more than ok for me, but I think it would be more intuitive if that option is present as a clear option in the objects settings.
I have tried adding this second extruder, I followed the instructios I found in the WIKI ecion here in github. I have my normal PLA profile, and I created a copy of that profile, but with the cooling turned on all time at 60% . After creating the second extruder and setting this 2 filament profiles, in the plater screen appears as expected to filament profiles, I guess the first one is suposed to be the one assigned to extruder 1 and the other one is the assigned to extruder 2.
But it doesnt work or at least the cooling is not activated . I know the extruders setup is working because in my first attempt, I forgot to put on Cero the retraction, and the defaul 10 mm caused a huge overextrusion when the extruders changed between objects. After correcting that, all worked OK, but the cooling is not activated. I hae atached the profiles and the stl plate
[Lightsaber.zip](https://github.com/alexrj/Slic3r/files/694533/Lightsaber.zip)

lordofhyphens, 2017-01-09:
The other thing I'd suggest (forgot to mention it before) is to pass it
through a post-process step that strips out the toolchange gcode. It might
not actually mean anything though.

My suspicion though is that Slic3r may be considering each extruder
together for time purposes; did your test change the min/max fan speeds for
the overhangs then (or told the fan to be "always on" with that profile?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2017-01-10:
One of the filament profiles has "keep fan always on" not activated and auto cooling activated with bridges fan speed set to 80%. The other profile (the one not being trigered), has keep fan always on activated, auto cooling not activated and bridges fan speed set to 80% too, and min fan speed set to 60%

lordofhyphens, 2017-01-10:
Damn, I was hoping that would work.

Tinchus2009, 2017-01-12:
So it look there is actually a bug here. Slic3r is takeing the extruder change, but it look like it is not acepting the different filament profile change

lordofhyphens, 2017-01-12:
I think this is related to #3664 then.

Tinchus2009, 2017-02-05:
Yes, look like it is something similar to #3664 . And no, cooling cant be controlled by modifiers, would be nice if it could.  This can be closed I think, I hace searched and actually what Im describing about support generation is already describe as support being created under bridges #3672

tullo-x86, 2017-03-11:
Hint: Anyone looking for selective support generation using modifiers should refer to (and follow) #2570.

Tinchus2009, 2017-03-11:
This is not about selective support generation. The report here talks about supports being built overriden by the option "dont support bridges"

tullo-x86, 2017-03-11:
My apologies; I found this issue when searching for a way to hint supportable volumes to Slic3r, and going by the screenshots, it seemed appropriate.