slic3r/Slic3r #3661: Modifier meshes suffer XY offset when using separate extruder

tjhinton, 2017-01-10:
**### Version**
v1.3.0-dev

**### Behavior**
Modifier meshes using a separate extruder are loaded with seemingly random x/y offsets, making it difficult or impossible to load them with correct spatial relation to a parent solid.  

1. open an stl (e.g. calibration cube) with 0,0,0 representing the center of the bottom face.
2. in the settings for that solid, open a modifier mesh for any other solid sharing a similar 0,0,0 bottom center.  
3. Set extruders for modifier and parent meshes to different values
4. observe the modifier mesh's seemingly random offset from the parent mesh's coordinate system.

Expected: modifier meshes should share coordinate system origins with parent meshes
Actual: meshes do not share origins

Note: there's a screenshot showing the bug in the zip.  

**#### STL/Config (.ZIP) where problem occurs**
[3-part-pawn_STL.zip](https://github.com/alexrj/Slic3r/files/696873/3-part-pawn_STL.zip)

tjhinton, 2017-01-13:
Has anyone had a chance to try replicating this?  Can I provide any additional information to help?

lordofhyphens, 2017-01-13:
Oh, modifier meshes default to extruder 1 because they aren't printable.
Maybe the entire object is being offset in the coordinates?

Could you add some screenshots from the interface? Both the object settings
dialog and the 3d pane. Maybe preview pane if you do something obvious like
change infill pattern or % to something easy to see?

tjhinton, 2017-01-13:
I found a slightly different way of recreating the random offset.  If you have a parent and modifier mesh, and you simply select them, sometimes the parent mesh is randomly offset.  I'll post screenshots of the necessary steps for me to get this to happen.  Basically, open an stl (we will call this the parent mesh), then open a modifier in the settings for that stl.  Select the parent, then select the modifier, then select the parent, then select the modifier.  You should see a random offset happen to the parent/modifier.  No action aside from selecting one then the other is required to cause this offset for me.  

[image]
[image]
[image]
[image]