slic3r/Slic3r #3662: g-code long saving time

fobetor, 2017-01-11:
Hi
 My question may be... obvious, but.... I have sliced an object (a big one), and the slicing took just 5 mins, but I noticed that the saving time of the gcode takes nearly 1 hour!!!... With other objects it takes a lot of time compared with the slicing time, but since the other objects I was sliceng were smaler I didnt bother about it....
 ¿Is this a characteristic of slic3r or it may be something to do with my computer ?

### Version
_Version of Slic3r 1.2.9

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

xoan, 2017-01-11:
Please, follow the [CONTRIBUTING](https://github.com/alexrj/Slic3r/blob/master/.github/CONTRIBUTING.md) guidelines.

fobetor, 2017-01-11:
version 1.2.9a
Windows x64 
¿Something more? It's just about saving time to gcode file

xoan, 2017-01-11:
Think in _what_ we need to reproduce it, and you find the missing things in your issue report.

bubnikv, 2017-01-11:
The G-code export has been significantly sped up since 1.2.9, but the
"avoid crossing perimeters" feature is still very slow if used on complex
objects.

fobetor, 2017-01-11:
Thanks bubnikv, I will try to disable it and test saving times

bubnikv, 2017-01-11:
But please use either the builds by @lordofhyphens or build the @alexrj master yourself, or youse the prusa3d builds. Please don't compare against the 1.2.9 version.

fobetor, 2017-01-11:
Thanks bubnikv!!!!! You are right, that is the answer. The time is reduced a 90% when disabling "avoid crossing perimeters".... 
BTW, another stupid question I cant find answer for... I have disabled the preferences option "background processing", BUT now.. I cant find a menu, option, key to process the files so I can preview them instead of saving directly the gcode (I need to previe since I rotate, scale etc the objects)
Thanks

bubnikv, 2017-01-11:
@lordofhyphens implemented a "slice now" button, but this has been then later replaced by @alexrj : by switching to the paths preview the paths are calculated automatically. The prusa3d fork uses the original idea of @lordofhyphens

fobetor, 2017-01-11:
ok. Thank you very very much.
Best

lordofhyphens, 2017-01-11:
@fobetor with the recent upstream builds, moving to the "Preview" tab initiates a reslice so you can preview. I'm generally okay with that workflow myself. 
Link: https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev

closed 2017-01-11 by lordofhyphens

fobetor, 2017-01-12:
And.. is there a 64 bit compiled version? (just in case). This version in x64 crashes ..... I am not good about compiling things...
Thanks
Fernando