slic3r/Slic3r #3663: Build fails

burbilog, 2017-01-11:
### Version

1.2.9-616-g8838763

### Operating system type + version

Ubuntu 14.04.5 LTS

### Behavior

Pulled today slic3r and now perl Build.PL fails:

Successfully installed Slic3r-XS-0.01
1 distribution installed
t/angles.t ..................... ok     
t/arcs.t ....................... ok     
t/avoid_crossing_perimeters.t .. ok   
t/bridges.t .................... ok     
t/clean_polylines.t ............ ok   
t/clipper.t .................... ok   
t/collinear.t .................. ok     
t/combineinfill.t .............. ok   
t/config.t ..................... ok   
t/cooling.t .................... ok     
t/custom_gcode.t ............... ok     
t/dynamic.t .................... skipped: variable-width paths are currently disabled
t/fill.t ....................... 90/92 
#   Failed test 'infill is only generated under the forced solid shells'
#   at t/fill.t line 300.
# Looks like you failed 1 test of 92.
t/fill.t ....................... Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/92 subtests 
t/flow.t ....................... ok   
t/gaps.t ....................... ok   
t/gcode.t ...................... ok     
t/geometry.t ................... ok     
t/layers.t ..................... ok   
t/loops.t ...................... skipped: temporarily disabled
t/multi.t ...................... ok     
t/perimeters.t ................. ok     
t/polyclip.t ................... ok     
t/pressure.t ................... ok   
t/print.t ...................... ok   
t/retraction.t ................. ok     
t/shells.t ..................... ok     
t/skirt_brim.t ................. ok   
t/slice.t ...................... skipped: temporarily disabled
t/support.t .................... ok     
t/svg.t ........................ ok   
t/thin.t ....................... ok     
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   

Test Summary Report
-------------------
t/fill.t                     (Wstat: 256 Tests: 92 Failed: 1)
  Failed test:  90
  Non-zero exit status: 1
Files=33, Tests=520, 42 wallclock secs ( 0.14 usr  0.04 sys + 42.98 cusr  0.73 csys = 43.89 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

A day or two ago everything was ok.

lordofhyphens, 2017-01-11:
Yeah, this is already captured via the build server.

lordofhyphens, 2017-01-11:
https://travis-ci.org/alexrj/Slic3r/builds/191046100 was commit that broke it. 

The test failure is likely spurious; @alexrj changed behavior for infill and the test wasn't updated to go with it. Not sure why he didn't see the test failure on his end.

closed 2017-01-11 by lordofhyphens