slic3r/Slic3r #3665: Feature request: Define outermost perimeter extruder

GFBsoul, 2017-01-12:
### Version
1.2.9 Bundled with Repetier Host

### Operating system type + version
Windows 10

This is a feature request.

Right now I'm printing something and find myself running out of filament in the right color. The thing I'm printing requires 4 perimeters and an infill of 25%. I extruder 1 for perimeters and extruder 2 for infill. 

What I would like to do is print infill and 3 of the perimeters with my extruder 2 and just print the outermost perimeter with extruder 1.

This feature would enable people to use cheaper filament for infill and some part of the perimeter and more expensive filament for the outermost perimeters.

bubnikv, 2017-01-12:
You would want to print the top and bottom surfaces (the skin) with the same extruder, right?

GFBsoul, 2017-01-12:
Yes, that's correct. All of  the skin with extruder 1, in my case.