slic3r/Slic3r #3666: Create option for gyroid infill

mrjohnc, 2017-01-13:
Please create gyroid infill that is the shape as graphene. This shape has been shown to be very strong as a 3D printed object and if used as an infill would presumably create very strong 3D prints

http://itsybitsymag.com/2017/01/11/mit-research-ten-times-strong-lightweight-graphene/

I printed a gyroid shape where all surfaces are 0.4mm thick (the size of my nozzle), I think this is the same as what the infill would be like, the model is 40mm x 40mm. It is very very strong, I think I could maybe stand on it and it would be ok, it weighs very little.

I got the model from here http://www.3dprintmath.com/chapters/6-Surfaces

[image]

Note: I have also created this request on the Prusa fork of Slic3r (I have a Prusa Mk2) https://github.com/prusa3d/Slic3r/issues/92

forReason, 2017-01-26:
seems nice but is it stronger than current infill like 3d honeycomb? seems rather comlex...

mrjohnc, 2017-03-09:
I created some test cubes to compare the strengths of infill, any ideas how to compare strengths? 

http://www.thingiverse.com/thing:2091011

lordofhyphens, 2017-03-09:
There is the crush test: use a press and slowly increase pressure until
something gives.

RafaelEstevamReis, 2017-03-09:
Poor man solution:
Put the object on a Scale and a cup on top of it and reset (zero) scale's
value.
Then put gravel, rocks, sand, water (caution, the cup can fall) until you
hear something crack.
Look at the scale.

lordofhyphens, 2017-03-30:
@mrjohnc any results from experimenting?

mrjohnc, 2017-03-30:
@lordofhyphens not really, I'm stuck on how to test fro strength. Any ideas that would be of suitable quality and repeatability?

lordofhyphens, 2017-03-30:
Crush test. If you are around a university with a mech engineering program
then they would have a good tool for measuring structural strength.

lordofhyphens, 2017-03-30:
Make sure you print with the same settings and the same roll of filament.

For your modeled boxes (assuming that the infill is modeled before it gets
to the slicer) perimeters very high and set infill to 0%.

Work out how much force it takes to deform and break the object.

Do it on both axes (along and against the layer lines).

lordofhyphens, 2017-03-30:
And weigh the boxes ahead of time.

ehybrid, 2017-04-08:
Even if it wasn't stronger, it should be print faster than 3d honeycomb due to the lack of corners, right?

NateTG, 2017-04-08:
>  Even if it wasn't stronger, it should be print faster than 3d honeycomb due to the lack of corners, right?

I don't think it matters.  Even if it did, there are simpler infill patterns (like cubic infill) that don't have corners either.

Shokman, 2017-04-10:
I was looking for a slicer that uses this infill after I saw this video:
http://news.mit.edu/2017/3-d-graphene-strongest-lightest-materials-0106

Here is the paper:
http://advances.sciencemag.org/content/3/1/e1601536.full
http://advances.sciencemag.org/content/advances/suppl/2016/12/30/3.1.e1601536.DC1/1601536_SM.pdf

They 3D printed several cubes and tested their properties. like you wanted to do.

I think that the results show that it is worth trying, but there is need for further research. If you are interested in doing something with this I might be able to get people involved from my university with access to equipment to test the printed parts properly.

supermerill, 2018-01-31:
@lordofhyphens 
>> Shouldn't be too bad. Download Slic3r-perl from wiki and put everything in
path.
Tried to compile the alexrj version of slic3r : windows fail (tried with the environment from silc3rPE: fail, your wiki : too much to do vs ubuntu vm). Fresh ubuntu 16.04 (version from the wiki)  fail 2 times (one because my vm disk has a bug, second time because it seems 16.04 has a wrong version of wxperl?)
Already lost a day on that (compiling is sloooow). 

I didn't find your "Slic3r-perl from wiki"

I'm out for now.

lordofhyphens, 2018-01-31:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows
@supermerill you mean that? Right there at the top of the Windows compiling instructions.  Header "Slic3r Perl"

supermerill, 2018-02-13:
boost fail to compile  ("don't know how to make link=static")
Finally succeed to compile on linux with a xubuntu 17.10.
Now, i have to convert some C++ to perl, and i'm not good at perl.
I have succeed to compile & test the gyroid infill, but i have some quirks with the ironing code.

I can make a gyroid pull request.
Do you want a buggy (the first top infill is missing, i'm not sure the gcode is well ordered) pull request for ironing? I can't use any more days to debug this. Maybe you can debug it easily (it works in the PE version, but the code base is diverging).

lordofhyphens, 2018-02-24:
Merged from work done by @supermerill. Thank you!

closed 2018-02-24 by lordofhyphens

chambm, 2018-03-14:
Nice work folks! Regardless of whether it's stronger or not, it's very cool looking. :) Perhaps I should subject it to the toddler jump test...

mhujsak, 2018-03-14:
Is this downloadable/usable now?  I was just thinking about gyroid infill today and looking for a way to do it myself - then I came across this thread.  I'm seeing builds dated 2-26 in the automated builds folder.

chambm, 2018-03-14:
Yep! [image]

mhujsak, 2018-03-14:
That is soooo cool!  You guys ROCK!

foreachthing, 2018-03-19:
@supermerill where can I post an issue concerning your wonderful infill? I do have sometimes problems with meshes not from CAD.

supermerill, 2018-03-21:
You can post it as a slic3rPE issue if you have a problem with slic3rPE.
You can send me a pm on reddit if it's not slic3rPE related.