slic3r/Slic3r #3667: Rotation of object direction is wrong

Louis-Cypher, 2017-01-14:
### Version
1.2.9 stable

### Operating system type + version
Windows 7

### Behavior
From my perspective the rotation direction of objects around X and Y and Z is wrong. I think the object should be rotated w.r.t the Printer COS not the other way round. According to the right hand rule Slic3r doesn't rotate the object, it rotates the printer

This is a bug

#### STL/Config (.ZIP) where problem occurs
Applies to any object

lordofhyphens, 2017-01-14:
Please include a few screenshots to show what you mean.

Patola, 2017-01-14:
I am curious to this bug because Slic3r actually follows the right hand rule (thumb → axis, fingers → rotation direction). Just tested here with an object, in all three axes. Slic3r 1.3.0-dev

Louis-Cypher, 2017-01-15:
Maybe I am just ill minded but this has disturbed me quite a while and meanwhile I just do the opposite of what intend to do and I am fine....

Here we go. Starting position:
[image]
 Rotate- X -> 90 -> Enter

[image]

Done

 From my point of view the beautiful lady should just be laying on her face, if she is the object that turns with respect to the printer, right?

Patola, 2017-01-15:
I just tested rotation in my slic3r. git master 586d95b28a8e473bdcbd959f1e37d215ae531df5

Loaded file:
[image]

Right-clicked on it, Rotate → Around X Axis... 90 (positive)

[image]

And this is the result:

[image]

Seems OK here. The Amitabha went face-down as expected. I have made an illustration of the right-hand rule, see if it was that:

[image]

Louis-Cypher, 2017-01-15:
That's really strange. In your example rotation is correct and we are using the same axis of rotation and the same rotation value.

Here is my config- bundle 
[Slic3r_config_bundle.zip](https://github.com/alexrj/Slic3r/files/706723/Slic3r_config_bundle.zip)
 
May be I made some weird settings changes?

Patola, 2017-01-15:
I doubt it. I know no configuration options which can change rotation settings....
Can you try and download and compile the latest version from git?

Louis-Cypher, 2017-01-15:
Sadly I can't on this computer since there is no compiler installed. I could do it on a linux computer but i expect no difference to what you experienced on your computer. Please remember this is a 1.2.9 stable issue.

lordofhyphens, 2017-01-15:
@Louis-Cypher https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-329 for a windows build.

1.2.9 isn't getting any commits to it, so any fixes would go into 1.2.10 (next stable) or 1.3.0 (current).

lordofhyphens, 2017-01-15:
@Louis-Cypher also, is the model available for download somewhere?

Louis-Cypher, 2017-01-15:
This issue occurs with any model. I know that there will be no commits to stable. Downloaded the build and will check

Louis-Cypher, 2017-01-15:
1.3.0. master as expected doesn't have the issue. This can be closed. Am I anyways supposed to report issues in 1.2.9 stable or only ones found in the most current development version?

lordofhyphens, 2017-01-15:
You can report issues from 1.2.9, but the first question someone will ask is "is it still a problem in 1.3.0-dev?" So be ready to answer ;) 

Things change from full release, often quite a bit.

Especially if there had been a reported issue with it before (my quick search didn't find any for this one). This is standard bug triage--if the symptoms don't show up in current -dev, then there's not much point in going further with it except for academic reasons (that I at least don't have time for).

If you want, you could probably nail down when the behavior changed with a git bisect routine. I am not going to bother. FWIW I never noticed that it was "wrong" myself with 1.2.9. If I had, it wasn't enough to trip my "maybe this is a bug" sense.

I provide the automatic windows builds so that people who use Windows can check 1.3.0-dev (or use it in general).

There's already some major improvements from 1.2.9 (especially around gap fill and infill overextrusion).

lordofhyphens, 2017-01-15:
I've updated the contributing guidelines to cover this, but who knows who actually reads this :(
https://github.com/alexrj/Slic3r/blob/master/.github/CONTRIBUTING.md

closed 2017-01-15 by lordofhyphens

Patola, 2017-01-15:
Just as a curiosity, by default Simplify3D also sports this "bug" (it follows opposite of the right-hand rule); actually, it does that because by default it flips the Y axis, but if you disable this flip in the preferences, it behaves just like Cura, slic3r and the rest of slicers (I happen to try and learn all of them).

lordofhyphens, 2017-01-15:
@Patola probably because people don't realize that on an i3 the y axis 0 is
all the way away from you?