slic3r/Slic3r #3668: Minor Bug in Documented Settings Section of GCode File - ",7" is Added

rf1k-mjh11, 2017-01-14:
### Version

Both 1.1.7 and 1.2.9

### Operating system type + version

Windows 7/64

### Behavior

This is a very minor bug (I assume):
In the generated GCode file, at the end, where all the settings are documented, you will find

_**filament_diameter = x.xx**_

The bug is that there is a ",7" appended to the diameter value ('comma 7').

For example:
filament_diameter = 2.86,7

I checked several older GCode files of mine.  Version 1.0.1 did not show this, but 1.1.7 and 1.2.9 both do.

A fellow user in a forum I am active in stumbled over the issue unknowingly.  I think he uses Windows 10.

#### STL/Config (.ZIP) where problem occurs

lordofhyphens, 2017-01-15:
Well, I can't reproduce this on Linux.

Can you check on 1.3.0-dev? https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-329#files

rf1k-mjh11, 2017-01-15:
I'll give it a try with the 1.3.0 version during the coming week.

mjh11

rf1k-mjh11, 2017-01-15:
I just tried it.  The bug is still there.  See the line with filament diameter.  It says: "1.72,7"

Here is a portion of the GCode file:

; generated by Slic3r 1.3.0-dev on 2017-01-15 at 20:43:04

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.35mm
; infill extrusion width = 0.35mm
; solid infill extrusion width = 0.35mm
; top infill extrusion width = 0.35mm
; support material extrusion width = 0.25mm
;
; filament used = 379.1mm (0.9cm3)

; avoid_crossing_perimeters = 0
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_connections_width = 0
; extrusion_multiplier = 1
; fan_always_on = 1
; fan_below_layer_time = 50
; filament_colour = #EFDE10
; **filament_diameter =** 1.72,7
; filament_max_volumetric_speed = 0
; filament_notes = ""
; first_layer_acceleration = 0
; first_layer_bed_temperature = 42
;

But it is not a serious bug.  It has obviously been around a long time and all my prints were ok.  
It could be an indication of some poorly initialized variable or similar, that could lead to occasional issues, but I don't think so.  Alex would know better if it affects anything.

mjh11

rf1k-mjh11, 2017-01-15:
Hi Alex, buongiorno,

I just tried the 1.3.0 development version from the link below. The 
result was the same.  See the GitHub comment section.

mjh11

lordofhyphens, 2017-01-15:
What I do know is that it doesn't show up on Linux :/ #thankswindows

lordofhyphens, 2017-01-15:
Could you post your config (File->Export config)

rf1k-mjh11, 2017-01-16:
OK, done.  Good thought.  The error aleady exists in the config file, at least in two versions.

[config_(1,3,0).txt](https://github.com/alexrj/Slic3r/files/709084/config_.1.3.0.txt)
[config_(1,2,9).txt](https://github.com/alexrj/Slic3r/files/709085/config_.1.2.9.txt)

It's already in the filament profile file:
[PTG_D175_Yel.txt](https://github.com/alexrj/Slic3r/files/709087/PTG_D175_Yel.txt)

I'll try removing that offending part from the filament profile file and start up Slic3r again.  But I can't do an actual print test.  All my nozzles are out being coated with an abrasive resistant, low-friction coating.  I expect them back in about 8 days.

mjh11

lordofhyphens, 2017-01-16:
Print test isn't the problem here. What I think might have been the case is that you had managed to get the extra characters in there and it was just propagating them to the output.

lordofhyphens, 2017-01-16:
I removed the extra offending characters and they aren't in the output. GIGO. 
I don't know why the characters aren't being shown in the box offhand.

closed 2017-01-16 by lordofhyphens

rf1k-mjh11, 2017-01-16:
## **ISSUE CLOSED** - This is not a bug  reason: see below.

OK, Slic3r doesn't mind having the ",7" removed - no surprise there.

I changed the filament setting of the newly edited profile and saved it under a new name: Still **no** unwanted garbage at the end of the filament diameter setting. - Good.

I can offer a hypothesis of what happened:
My profile was generated off of one I received with my printer.  This profile seems to have had the ",7" in it.  Since it is not a valid number, Slic3r ignores it and puts the valid number in front of it.  Even if you change the filament diameter in Slic3r, the garbage at the end of the number is not removed or touched, and lives on forever.
I just tried it by taking the edited profile file, adding ",8" to the end of the number, slicing a part, and, no surprise: the filament diameter reads 1.725**,8**

How the comma-seven got there in the first place?
My theory: The printer manufacturer is German,  Germans use the comma in place of the decimal point, normally.  I suspect someone there edited the first profiles by hand and added "comma-seven" in place of "point-seven".

## **ISSUE CLOSED** - This is not a bug

Sorry to have wasted your time.

mjh11

lordofhyphens, 2017-01-16:
No worries, glad it was sorted out.