slic3r/Slic3r #3673: support is too firmly attached to base

wpwrak, 2017-01-16:
### Version

commit 1f920dc8470a42647db654e7d984e522a93538d2

### Operating system type + version

Ubuntu 16.10

### Behavior

When printing a simple design (link to STL below) with a flat base and a small opening, using "honeycomb" (but "pillars" is not much better), Slic3r places a solid block of material in the opening, and attaches the block just as firmly to the base as anything else. To allow breaking off the support, the link to the base should be much weaker.

#### STL/Config (.ZIP) where problem occurs

STL:
http://downloads.qi-hardware.com/people/werner/slvs/box.stl
SolveSpace model that created this STL:
http://downloads.qi-hardware.com/people/werner/slvs/box.slvs
Config:
[bug-config.zip](https://github.com/alexrj/Slic3r/files/709577/bug-config.zip)

lordofhyphens, 2017-01-16:
Looks like part of it is that the top interface is overwriting the bottom interface layers to me.

Because they're solid, it produces a solid interface on the bottom. I also wonder if it is properly enforcing a bridging gap from the bottom contact area. The preview doesn't seem to indicate that is happening.

lordofhyphens, 2017-01-16:
@wpwrak could you give this a try? Replace the SupportMaterial.pm file in your Slic3r-dev directory with this one.

It's a dirty hack that omits the bottom layer of support for the bottom interface layer.
[patched_support.zip](https://github.com/alexrj/Slic3r/files/709602/patched_support.zip)

lordofhyphens, 2017-01-25:
I suspect that this is related to #3674. 

@wpwrak I would very much appreciate it if you could indicate if the dirty hack worked.

bubnikv, 2017-01-27:
The bottom of the support has to be extruded with a bridging flow and there
shall be a gap guaranteed between the top of the object and the bottom of
the support. There is no way you can easily hack it with the current
support. I am putting the final touches to the brand new supports code in
the prusa3d fork. You may try the supports there if you are able to build
Slic3r from source. The binary builds will be released soon.