slic3r/Slic3r #3675: Unable to build with non-standard dynamic boost libraries under Linux

Multimath, 2017-01-17:
### Version
Trying to build git commit : 1f920dc8470a42647db654e7d984e522a93538d2
(since this is a build issue, I cannot follow the "guidelines for contributing" instructions to find the version number, something you should probably fix in the instructions, there --- it's not trivial to find the version number from the codebase.)

### Operating system type + version
Any Linux without boost libraries installed in system, /usr/local, or other places searched by default

### Behavior
* When attempting to build Slic3r from git using a non-standard dynamic-only boost library installation, configuration of ./xs fails no matter what the environment variables BOOST_{DIR,INCLUDEPATH,LIBRARYPATH} are set to. This is caused by lines 91-97 of ./xs/Build.PL explicitly ignoring them when calling check_dir(...), and lines 102-137 subsequently assuming that the user wants static linking. The user is then presented an error message which does not help him understand the problem.

Either the error message should explain better, or Build.PL should be fixed to handle this case.

(This is not a new feature request.)

And... thanks for all the great work! It **is** appreciated!

dmopalmer, 2017-02-20:
Also: the error message should refer to BOOST_LIBRARYDIR instead of BOOST_LIBRARYPATH

alexrj, 2017-03-18:
@Multimath, how did you install Boost libraries? Do you have system *and* thread *and* filesystem (all three are required).
Can you try with current master? We improved Build.PL.
Thank you!

Multimath, 2017-03-19:
I compiled Boost from source in an non-standard directory; after patching ./xs/Build.PL for my own unusual configuration I succeeded in building (of course I have to update LD_LIBRARY_PATH accordingly to run).

The bug report was only FYI. In general, under Linux you will get a wider range of configurations; especially in work situations people can sometimes be constrained to use old or stripped-down system versions and end up having to improvise accordingly.

I will try to review the fix in master when I get some free time for it. Sorry not to be able to immediately help you there. And thanks!

alexrj, 2017-05-16:
Thank you @Multimath, this was fixed! The easiest way is to supply BOOST_DIR to the root boost compilation directory.

closed 2017-05-16 by alexrj