slic3r/Slic3r #3676: print speed comes to crawl when 1st layer height is set to 0.3333

pflugj, 2017-01-20:
### Version
1.3.0-dev
slic3r-master.2017.01.16.332.1f920dc

### Operating system type + version
Windows 10
### Behavior
When using layer height of  0.3333 (all layers) and all speeds are set to auto.
The following happens
1st layer is OK!
as from the 2nd layer the print speed comes down to a crawl
here is a part of the gcode:
G1 Z0.667 F7800.000
G1 X-1.232 Y13.882 F7800.000
G1 E2.00000 F2400.00000
**G1 F4.88486**

* _Steps needed to reproduce the problem_
Set 1st layer height to 0.3333 and slice

* _Expected Results_
calculate the right speed?

* _Actual Results_
print speed comes down to a crawl from the 2nd layer

  * 
[config.zip](https://github.com/alexrj/Slic3r/files/718576/config.zip)
[Uploading config.zip…]()

_Screenshots from __*Slic3r*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[config.zip](https://github.com/alexrj/Slic3r/files/718576/config.zip)

lordofhyphens, 2017-01-20:
So it is less than 30mm/s?

pflugj, 2017-01-20:
If I am not completely wrong, after the layer change to layer 2
the G code contains the following:
G1 Z0.667 F7800.000
G1 X142.451 Y12.773 F7800.000
G1 E2.00000 F2400.00000
**G1 F69.1641**
G1 X139.557 Y14.288 E2.17067
G1 X136.480 Y15.385 E2.34134
etc.

...and I think **G1 F69.1641** 
is ~69 mm/per minute isn't it?

pflugj, 2017-01-20:
when trying different stl I get different results, but all very slow (almost not noticeable)
G1 Z0.667 F7800.000
G1 X-8.875 Y10.658 F7800.000
G1 E2.00000 F2400.00000
**G1 F5.56848**

lordofhyphens, 2017-01-20:
It's probably autospeed acting up. There is probably some slow part on that
particular model that rippled back to slow everything else down (trying to
print with constant pressure).

I think that autospeed needs some debug/verbose commentary as to why it is
setting that speed.

bubnikv, 2017-01-20:
IMHO the auto speed feature does not make sense the way it is implemented right now. You will get a more meaningful results if you just limit the acceleration / deceleration of the extruder in you firmware.

The auto speed feature in Slic3r knows nothing about the acceleration / deceleration ramps in the firmware. The auto-speed has to be optimized together with the acceleration / deceleration profiles of the firmware, otherwise one will break the other. It is much more complex than it seems.

pflugj, 2017-01-20:
I understand that auto speed is very complex.
I was just reporting that I noted a possible bug ;-)

lordofhyphens, 2017-01-20:
Yep, you did right.

As a workaround for you though, turn off autospeed ;)

pflugj, 2017-01-20:
Thanks, will do so ;-)

alexrj, 2017-03-04:
After the recent changes, autospeed is calculated on per-layer basis. I tested your config with a generic cube and I confirm speed does not go below 69.7mm/s for layers >= 2.

Could you please test with current master and confirm it's fixed? Thank you!

lordofhyphens, 2017-03-11:
This looks fixed to me. I tested http://www.thingiverse.com/thing:531497 with the provided config and the speed got as slow as 8-9mm/s (looks like at higher layers where there are small perimeters). The majority of the speed changes are over 70mm/s. 

Going to close for now @pflugj. If you still see issues, please let me know so I can reopen.

closed 2017-03-11 by lordofhyphens

pflugj, 2017-03-12:
Good morning.

I am sorry but I still think there is an issue ;-(

I used the attached stl and as before all automatic speed settings inSclic3r

All Ok here and print time in Repetier is ~3h...

and then I tried version 2017.03.11.476 which was the latest at the time 
I tested.
I can hardly keep up with the new versions pouring in ;-)

This time I used alayer height of  0.1 mm

At z=1.15 it started to slow down.
This time not to a crawl but speed went down significantly
I looked at the G-Code and found this:
G1 Z1.150 F7800.000
G1 X-7.128 Y-11.629 F7800.000
*G1 F438.611*
G1 X-6.520 Y-11.922 E0.01849

and Repetier reported a print time of *11h... +*

compared to Slic3r 1.2.9
G1 Z1.150 F7800.000
G1 X-6.876 Y-11.756 F7800.000
G1 X-6.161 Y-12.079 E0.02148 F2759.121

best regards from Namibia
Jurgen

lordofhyphens, 2017-03-12:
You can just use the short hash. It's what it is called on the build server.

Some of this is going to be unavoidable because of the widths and flows
used at low layer heights. If you fix your extrusion width to something
constant it will probably behave better.

pflugj, 2017-03-12:
I never played around with the extrusion width and left them always on "auto",
except for the 1st layer.
I always assumed that slic3r would know do a much better job than me.
But I will try that and see how that turns out.

FWIW I also had a look at the Slic3r Prusa edition 1.33.8  and there the preset shows that:
ALL extrusion width are set to 0.45 on a 0.4mm nozzle
Maybe because of the exact same reason?

lordofhyphens, 2017-03-12:
Slic3r auto width prefers to produce wider traces at low layer heights.