slic3r/Slic3r #3678: Bridge settings, an idea for a new feature

dobrzann, 2017-01-22:
this is a new feature request.

When the bridges are too long, lines are able to connect each other.
So maybe add a new function "Bridge settings"
-Bridge pattern angle
-Bridge pattern spacing (like in support settings)

is it possible?

bubnikv, 2017-01-22:
Frankly I don't understand. Would you please draw a picture and give more
detail?

dobrzann, 2017-01-22:
I mean, option to change spaces between bridge lines
[image]

landodragon141, 2017-01-22:
I'm pretty sure this does that.
[image]

dobrzann, 2017-01-22:
I think, you can change only extrusion width in this way (except extrusion width settings). Space between lines is the same. 
I think about something like "interface pattern spacing" in support options.

lordofhyphens, 2017-01-22:
What the bridge fill ratio does is extrude less plastic for the same traces.

jkoljo, 2017-01-23:
Yep, nowadays hot plastic from previous extrusion line can stick to nozzle, ruining the nice bridge. It would be great to be able to extrude every second line. And when those are done, do the rest, so lines have had time to cool down

dobrzann, 2017-01-23:
is it a problem to add this feature?

lordofhyphens, 2017-01-23:
@dobrzann it isn't necessarily a problem to add it in code, but we prefer to understand the problem  (and everything around it) before jumping to a solution.

Often what people ask for is not actually what they want. ;)

forReason, 2017-01-23:
I believe what he wants is the folloing:
currently the nozzle will bridge one line of plastic and continue with the next line beneath it.

This causes the first line to stay heated and "melt down".

he wants to print the first line, then let out one or to to and bridge the next line. Later one add the skipped lines so the plastic had time to cool down.

not sure tho if this would work and/or be useful

lordofhyphens, 2017-01-23:
I'd be worried about anchoring the bridge for those filler lines.

dobrzann, 2017-01-23:
> Only print every Nth line of the bridge so the nozzle wont come near previously printed lines and ruin them.
    Continue with the next layer building onto the already hardened lines

And this is exactly what I mean. But It can be Nth line of the bridge or defined space between lines, like "interface pattern spacing" as I mentioned before

Tinchus2009, 2017-01-24:
I understand what @dobrzann is asking for but, I say this with respect, @dobrzann wouldnt be better t have your bridging settings properly calibrated? With actual setting and proper calibrating (good bridge speed calibration, a good layer cooling and the correct bridging extrusion value) U can produce HUGE bridges (I have posted in thingiverse a bridging test of 15 cms long, with no sagging at all). 
So the problem I think is not a setting in slic3r but  correct tunning for bridging.

kefir-, 2017-01-25:
I agree that proper calibration is important, but I think different nozzles can cause different issues. I suspect smaller nozzle sizes can cause more trouble for bridging, and different nozzle designs may also have some impact on how the molten flow leaves the nozzle. And it'd be interesting to have some way to experiment further when no changes of speed, temperature and flow rate have managed to stop an otherwise well calibrated printer's nozzle from mopping up the previous bridge strand, for example.

lordofhyphens, 2017-01-25:
Thing is that the bridging flow is supposed to leave a good surface to
print on.

If you widen the space between the traces, you at least need another bridge
flow in the opposite direction that has the tight spacing in order to print
on it.

I can probably throw together a short hack that will do something like this
for people to experiment with, but I personally don't believe that it is
going to be a worthwhile endeavor.

Tinchus2009, 2017-01-25:
i agree about noozle design and its effects on printing but then we will be
talking about how to solve through software a hardware fedign problem. and
to me, bad dezigned/cheap hardwared has its prize. on the other hand, the
idea being proposed as a solution willbringnew problems: if 1 bridge fails
there will be a very nice hole in the surface. also, for haveing a decent
surface finish afyer a bridged area you need at least 3 layers.  with this
idea that number will for sure be increased.

dobrzann, 2017-01-28:
> I can probably throw together a short hack that will do something like this for people to experiment with, but I personally don't believe that it is going to be a worthwhile endeavor.

Please do it if you can.
Maybe it worth a try.

bubnikv, 2017-02-02:
I consider the idea to be sound. One cannot say whether it will work until one tries it.

I actually had the same idea some months ago and it is somewhere at the bottom of my veeery long list.

> if 1 bridge fails there will be a very nice hole in the surface. also, for haveing a decent surface finish afyer a bridged area you need at least 3 layers.  with this idea that number will for sure be increased.

The idea is to put down every 2nd layer and then to press the intermediate threads at the same layer in between the already placed threads. Maybe the 1st set of threads could be printed with the bridging flow with a bit less than twice the spacing, and the 2nd set of threads with a bit less material and pressed in between the already laid threads. My hopes are, that one would actually get a better filled bridge surfaces.

dobrzann, 2017-03-24:
I see the topic died :-)

@lordofhyphens 
```
I can probably throw together a short hack that will do something like this
for people to experiment with, but I personally don't believe that it is
going to be a worthwhile endeavor.

```

Is there any chance you will add this function for the tests?

sachel85, 2018-01-11:
Anyone still working on this?

lordofhyphens, 2018-01-11:
I am not working on it. I am too busy.

@sachel85 if you want to take a run at it be my guest.