slic3r/Slic3r #3680: [Feature Request] - Set-up tutorial (with pictures)

forReason, 2017-01-24:
I would like to see some get started/Set-Up tutorial for Slic3r Settings.

Reason being is that I see many threads about how to calibrate the Printer (slicer settings). Also I see many bad calibrated Printers. By that I mean settings in the Slicer like Printspeed, Layer height, Retraction, Temperature, Flow and so on.

If there would be a button to a Wiki page or even a "guided wizard" that would kill a lot of those threads I think. Hopefully it would be the first place that the user approaches. The usability and friendlyness to beginners has to be massively improved from my point of view. Its a factor that is often overseen by experienced Makers.

What I would request would be a standardized Process for basic setup which could be like the following:

1. Choose some standard settings for your filament and Printer type (Bowden/Direct)
2. Print an Vase Object (with provided stl) which increases Temperature over the layers upwards
3. Analyze the Print and choose the right temperature
4. Do the same with Printspeed
5. Settings for max Overhang, Bridging and support Material
6. Maybe other Settings

This is my personal process for setting up a new filament I get. Maybe there are better processes out there. Especially for the settings changed during the Layers a generator would be very good which could be combinated with a guided wizard.

A Wizard could have the following procedure:
1. Some short explanation text
2. Choose your Printertype [Bowden/Direct]
3. Choose your Filament (Presets like PLA/ABS and maybe other common Filaments / Custom for custom Settings)
4. Explanation with picture how to analyze the object with button "generate g-code"
5. enter your result
6. Continue with other Settings

Patola, 2017-01-24:
The Manual does not already have this with the Configuration Wizard screenshots? http://manual.slic3r.org/first-print/configuration-wizard

forReason, 2017-01-24:
Maybe I have to Precise it A little. 

Every Filament is different therefore it is important to find the ideal settigns for the different Filaments. You may print with the provided temperature of the manufacturer but if you want to have good results you have to find the best settings for your particular filament. for this reason a testing process is required

please have a look at the following picture [http://www.thingiverse.com/thing:1548697](http://thingiverse-production-new.s3.amazonaws.com/renders/96/e3/5c/55/3d/491730c03f3cd431ee343f3425c278dc_preview_featured.jpg) you can see: any above 190° gives bad results so will any setting under 190°

lordofhyphens, 2017-01-24:
The wiki is, as far as I know, open to edit. Open up a page and draft out
some step by step instructions.

We can canvass for some volunteers to test the instructions. Once we know
that the instructions are repeatable and useful, we can look at getting it
into code as a wizard.

forReason, 2017-01-25:
Will do something later in the year :D

bubnikv, 2017-01-27:
@forReason Great ideas, volunteers are always welcome :-)

forReason, 2017-01-27:
I decided to reopen the "Issue/Feature" request as I am currently on the Work. I have seen that a Wikipage already exists and started editing it because the page was fairly poor.

Anyone wanting to contribute or check the edits is welcome to do so

forReason, 2017-01-27:
okay now I need reviewers and testers to then come up with some neat ideas and scribbles of possible gui solutions

https://github.com/alexrj/Slic3r/wiki/Calibration-of-Slic3r

lordofhyphens, 2017-01-27:
Let's get the process worked out completely and then worry about GUI solutions. At minimum then there will be good instructions up for people to work with.

forReason, 2017-01-27:
kinda worrysome that it took 5 years to get instructions going 😅

lordofhyphens, 2017-01-27:
It's been on the to-do list, but there's also a plethora of "how to
calibrate, etc" on the reprap wiki. Personally I'm not convinced it needs
to be rolled into the slicer, but I've been wrong before.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

forReason, 2017-01-27:
well It soesnt have to be rolled directly into slicer but there could be a lot of neat tools like generate gcode with gradient over layerheight for easier calibration.

From my point of view it is good to have people out there doing a "how to" bot not a single documentation out there i have ever seen was complete. Its really about finding the breadcrumbs at the beach.. There just has to be at least something like a apropriate wiki page for the new user.

lordofhyphens, 2017-01-27:
@forReason Thing is that a lot of the "recommendations" in 3d printing are cargo cult science. Someone did 5 things, thought one of them worked, and then spread it around (despite not properly experimenting and controlling variables). http://www.wastedtalent.ca/node/3044 
It's also compounded by lack of uniformity in equipment and measurements and environmental things that often aren't considered. 

So, yeah, "WorksForMe".