slic3r/Slic3r #3681: [Bug] Slic3r crashing when setting bed size

forReason, 2017-01-24:
### Version
lordofhyphens / Slic3r / Slic3r_Branches / 1.3.0-slice_statistics-337

precompiled version

### Operating system type + version
Windows 10

### Behavior
Slicer will crash when setting a different bedsize. I think it is a null error with live preview.

steps to reproduce the problem:
1. Open slicer
2. Go to printer settings
3. Bed size
4. Delete the first number of the preset 200 in orter to type another number like for example 300 -> there will stand 00 - Slicer crashes

[image]

lordofhyphens, 2017-01-25:
Good catch. The actual bug is that the input validation code doesn't properly type munge if you give it "00" or "0000" or similar.

forReason, 2017-01-25:
Please close the issue when Solved. for some reasons I cant run Slic3r from the branch.

Also this may be a bad question but is there any outlook on when Slic3r 1.3 will get released? Because its now 2 year since the last release and I love the new features like instant re-slicing and the preview and so on.

lordofhyphens, 2017-01-25:
It will be closed when the fix is merged.

lordofhyphens, 2017-01-25:
@forReason I did test it before I committed. 

https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-fix_3681_zero_bed_shape-338 is the branch with the fix. The slice_statistics branch is a placeholder until that gets merged and so will not see this fix.

PR #3683 (2017-02-07): Catch input that resolves to 0 in bed shape dialog
Fixes #3681
commit c557a005 (2017-02-07): Fixed input to bed shape dialog to catch everything that resolves to 0. Fixes #3681 (#3683)
closed 2017-02-07 by lordofhyphens