slic3r/Slic3r #3682: processing triangulated mesh hangs

faceless105, 2017-01-24:
### Version
1.2.9

### Operating system type + version
Linux Mint

### Behavior
I'm working on a 3d print and I'm slicing a print into smaller sizes for my printer. I'm working on the first layer here: https://drive.google.com/open?id=0B-uDAjUUkNcdSFBsbFptZnRXM00
When I rotate it around the y axis 90 degrees and cut it about half way up, I get the two pieces, but it seems stuck on "Processing Triangulated Mesh". I'm not sure what's causing it, I've checked that linux is up to date and that slic3r. I've rebooted and tried a few times. So far I keep getting the same result.

lordofhyphens, 2017-01-24:
Pull and build 1.3.0-dev according to the wiki instructions. Let us know if
it is still a problem.

Please upload your exported config (you should be able to upload to github
a zip file) as well.

I can't reproduce without that information.

lordofhyphens, 2017-01-25:
As I thought it might, slices just fine with my settings on 1.3.0-dev.

I suspect that this is actually the (fixed) bug #3613.

Closing. If you can reproduce on current 1.3.0-dev or provide your exported config.ini (File->Export Config), I will reopen.

lordofhyphens, 2017-01-25:
Also, "pics or it wasn't fixed" ;)
[image]
[working-config.zip](https://github.com/alexrj/Slic3r/files/728602/working-config.zip)

closed 2017-01-25 by lordofhyphens

faceless105, 2017-01-25:
I appreciate the help with this. Moving to 1.3Dev fixed all my problems.