slic3r/Slic3r #3684: Crash when slicing a simple model

VanessaE, 2017-01-25:
### Version

git commit 586d95b28a8e473bdcbd959f1e37d215ae531df5

### Operating system type + version

Debian 8.6

### Behavior

Crash when slicing the "300" model linked below.   That file is a copy of the "orig" file also linked below, scaled up to 300% via Blender.  The crash also happens if I load the "orig" file into Slic3r and scale it there, but only if I scale it larger than about 205% (I did not test whether the same is true for scaling via  Blender).

The config at the time was my "60 mm/sec, 0.1mm layers" print profile (config.ini.zip below); if I switch to my "60 mm/sec, 0.2mm profile" (config-2.ini.zip), it doesn't crash.

```
vanessa@rainbird:~/RepRap/Slic3r-alexrj$ cat ~/Scripts/run-slic3r.sh 
#!/bin/bash
eval $(perl -Mlocal::lib)
perl ./slic3r.pl $*

vanessa@rainbird:~/RepRap/Slic3r-alexrj$ ~/Scripts/run-slic3r.sh 
Use of uninitialized value in subroutine entry at /home/vanessa/RepRap/Slic3r-alexrj/lib/Slic3r/GUI/Plater.pm line 1228.
*** Error in `perl': double free or corruption (fasttop): 0x00007f48e8048a40 ***
/home/vanessa/Scripts/run-slic3r.sh: line 3: 22033 Aborted                 perl ./slic3r.pl $*
```

#### STL/Config (.ZIP) where problem occurs
original-size file: [IDIC-orig.stl.zip](https://github.com/alexrj/Slic3r/files/729175/IDIC-orig.stl.zip)
scaled to 300%: [IDIC-300.stl.zip](https://github.com/alexrj/Slic3r/files/729176/IDIC-300.stl.zip)

These files were derived from http://www.thingiverse.com/thing:875580
(I split it apart for multi-color printing, fixed non-manifold polys, then scaled the result)

60 mm/sec, 0.1 mm layers: [config.ini.zip](https://github.com/alexrj/Slic3r/files/729187/config.ini.zip)
60 mm/sec, 0.2 mm layers: [config-2.ini.zip](https://github.com/alexrj/Slic3r/files/729190/config-2.ini.zip)

lordofhyphens, 2017-01-25:
Just a FYI,  you don't need the eval() with any -dev build of Slic3r newer than d7f3393c262ed53e475764497a76e2c0ea90959a or so. In fact, it may be doing weird things to your environment at this point. 

It probably isn't the cause of this (I haven't had the time to check myself yet with your model and configs), but I thought you should know.

VanessaE, 2017-01-25:
If I drop that line, it throws this error, 

```
Encode::Locale version 1.05 required--this is only version 1.03 at /home/vanessa/RepRap/Slic3r-alexrj/lib/Slic3r.pm line 80.
BEGIN failed--compilation aborted at /home/vanessa/RepRap/Slic3r-alexrj/lib/Slic3r.pm line 80.
Compilation failed in require at ./slic3r.pl line 16.
BEGIN failed--compilation aborted at ./slic3r.pl line 16.
```

lordofhyphens, 2017-01-25:
I'd probably make sure that you don't have an eval() hanging around
somewhere in your shell profile (.bashrc or .profile) and purge the source
directory and rebuild.

Odds are Slic3r's libraries and dependencies were getting placed in ~/perl5
instead of ./local-lib

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2017-03-18:
I can reproduce this as of current master.

alexrj, 2017-03-18:
Fixed! Thank you.

closed 2017-03-18 by alexrj