slic3r/Slic3r #3685: [Feature Idea]

forReason, 2017-01-25:
I just had a neat Idea but don't know if this would be apreciated by the community

### Version
future

### Behavior
I had the Idea to add a color attribute to Filament.

Rather than just beeing green in the plater you could instantly see which filament settings you have set

For example PLA beeing green,
ABS beeing gray and so on.

this might help preventing Setting issues but might as well be totally useless

lordofhyphens, 2017-01-25:
There already is a color attribute for filament and the profile box will
show it.

However the visualization (in my mind, correctly) uses different colors for
different types of lines (support, infill, perimeters). Permitting custom
colors leads to confusion in the interface.

I would not work on or merge such a feature without careful consideration
in its definition and intended use.

gege2b, 2017-01-25:
Maybe using the custom color in the 3D plater then ? (I think that's the point of @forReason post)
Each object would be colored by his filament's color

I agree that 3D preview must keep the current coloring scheme

lordofhyphens, 2017-01-25:
@forReason Annotated screenshots or UI mockups would be very helpful in understanding what you mean.

forReason, 2017-01-25:
will do some in the evening.
I agree with gege2b to add the custom colors to the plater only.

forReason, 2017-01-25:
I have just seen, It is not possible (yet) to just have different sets of settings and "attach" them to the objects... I am just too used to simplify 3d o.O (because of this particular feature)

lordofhyphens, 2017-01-25:
With filament and extruder settings, those are handled by assigning a
filament + extruder to an object.

I agree that being able to override with a profile for an entire
object/part is a useful feature to have. It shouldn't be too crazy to add
either.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-01-25:
@forReason yes, create a separate feature request (first search to make sure it hasn't come up before)

alexrj, 2017-03-01:
Coloring the 3D toolpaths preview using filament color is something I thought about. (Has to be configurable)