slic3r/Slic3r #3687: Feature Request

artbody, 2017-01-26:
### Behaviour
since today each layer is the same hight. Printing a fine structured  surface  needs a lot of time

### this is a new feature request?

this idea will speeding up print time 2 or 3 times
if we print with layer hight i.e. 0,06 mm  this needs a lot of time but gives a nice surface
so my idea was to print the wall zone with 0,06 mm 3 Layers in hight and i.e. 3 lines width and then print one Layer with 0,18mm
ooo  ------------------------- ooo
ooo  ||||||||||||||||||||||||| ooo
ooo  ------------------------- ooo
o = 0,06mm
| =  0,18mm

> wallsubdivide by 2x2 3x3 up to  6x6 

i'm building a printer (based on Prusa i3) with 90cm x 90cm x 40cm printing area.
so in this dimension such a feature will increase printing speed dramatically

forReason, 2017-01-26:
please update your title like
`[Feature Request] - Layerhights`

forReason, 2017-01-26:
I guess can be closed

You have to activate `Mode: Expert` Under `File/Preferences`
[image]

lordofhyphens, 2017-01-26:
See both combined infill setting and the adaptive slicing pull request by @platsch.

Closing.

closed 2017-01-26 by lordofhyphens