slic3r/Slic3r #3688: [BUG] - Infill Weired Behavior

forReason, 2017-01-26:
### Version
Latest build

### Operating system type + version
Win 10

### Behavior
Cant explain whats going on. Just look at the screenshot. Infill gets WAAAAAY too big. Only the first layer(s)

What Settings have I used?
Nozzle .5mm
Perimeters: .06 mm thick
Infill: 50%, every 6 layers

### STL I noticed with:
[3DBenchy.zip](https://github.com/alexrj/Slic3r/files/732463/3DBenchy.zip)

lordofhyphens, 2017-01-26:
Post a zip of the exported config.ini (File->Export Config).

Like the contribution guidelines asked.

forReason, 2017-01-26:
Neems to generally apply for low layer hights.

It seems like this issue is not related to print infill every x layers.

(theortetically) I should be able to print up to 0.0125mm layerhights with fullsteps. not to mention what happens with slicer then
before posting Config. ini i will experiment a bit shortly to get a better understanding of the bugs behavior.

forReason, 2017-01-26:
Okay. I found some interesting behaviors

the following Settings I have tried

|Z-Height|Infill Behavior|
|-|-|
|10mm|normal|
|1.25mm|normal|
|0.5mm|normal|
|0.4mm|normal|
|0.3mm|normal|
|0.352mm|normal|
|0.125|bigger than normal|
|0.1mm|FAT|
|0.06mm|HUGE|
|0.0125mm|HUGE|
|0.01mm|Size normal but infill gets solid 100% eventhough 50% are set|
|0.001mm|size and infill as above + edges get very noisy|

sizes from smaller to bigger:
small - normal - bigger than normal - FAT - HUGE

[settings and test stl.zip](https://github.com/alexrj/Slic3r/files/732594/settings.and.test.stl.zip)

lordofhyphens, 2017-01-26:
In general, if you are reporting a bug and don't provide a way for someone else to easily reproduce the problem, your issue will be tagged as needing more information and ignored until the information is provided. We have better uses of our development time than attempted mind reading. 

Not intended to be rude or dismissive of your report. This is just how it is.

I bet this has something to do with Slic3r's choice of sparse infill width.

forReason, 2017-01-26:
my apologies i thought the testfile and the wrote down config would be enough.

From my point of view its concerning that it already starts at 0.1 mm

lordofhyphens, 2017-01-26:
Why is it a problem, by the way? That is something I am having problems
understanding from your report.

It looks to me like Slic3r is trying to satisfy some minimum flow for
sparse infill.

forReason, 2017-01-26:
at 0.15 this is not a problem
but lets print with .06 mm:
[image]

you will have a cross as infill. good luck getting the top layers down nicely
Or even less:
[image]

You may have all support material from the layer in a single blob in the middle - I don't even want to try that with my printer as it may or may not fail critically. 

However I sucessfully printed with 0.01mm layerheight in the past with 100% infill (where this issue wont occur)
This was for tiny parts but seriously, you dont want to print something bigger than 1x1x1cm with 100% infill and 0.01mm layerheight as it will take weeks

lordofhyphens, 2017-01-26:
Yeah, that other picture what what I was wondering about; there's been related bugs that were fixed to this. 

Thanks, although I haven't seen this myself (and I print some pieces at 0.0475). I'll cross-check this tonight against my own settings and see where the issue could lie.

Have you played around with the "Infill every layers" setting? It's quite useful for speeding up prints at absurdly low layer heights.

lordofhyphens, 2017-01-26:
> my apologies i thought the testfile and the wrote down config would be enough.

This is why we ask for a complete exported config, test model, and picture where possible. It's very important that I replicate your setup as close as I can, and there's a lot left unsaid with the way you wrote it. The underlying assumptions baked into one of my configurations will likely be different than one of yours. Those different assumptions often are critical to reproducing the symptoms. 

It's also more work for me to set up a configuration. Both attributes make investigating more difficult (and thus less likely for me to bother if I have something else to do).

An exported config, a sample test, and a picture for what to look for (which you have provided, thank you) solves both of those issues.

lordofhyphens, 2017-01-26:
Your infill overlap is set very high (55%) in my experience. You may want to reduce it and see what happens.

forReason, 2017-01-27:
I have played with the infilll Every X Layers to Provide support for a maker here I first (in the First Post) thought its coming from there but I had it set back to infill every layer and it still was there. I don't know if the bug was already there back in the day when I printed really thin. 

For the overlap: im using standard slicer settings for the testing. I'm working with 1.2.9 on a different pc but I can test it tomorrow with (no overlap I think?) and maybe slicer 1.2.9 maybe it helps to know if the bug is actually new

lordofhyphens, 2017-01-27:
Probably related to #2599 as well.

forReason, 2017-01-27:
May or may not. Sounds like similar Eventhough I can't find issue's with top and bottom layers (as seen in the screenshots

lordofhyphens, 2017-01-27:
Yeah, I just reproduced this on my system. 

There's a few things going on here. 

1.  `width = ((nozzle_diameter*nozzle_diameter) * PI + (height*height) * (4.0 - PI)) / (4.0 * height);`
This is the "default" width calculated for everything when '0' is set. It's then checked to be within some min/max value.
1. Internal infill (like what you are seeing) has no cap on the internal width (see #2599). 
1. Thus, with your settings, (0.5mm⋅0.5mm) ⋅ pi + (0.0125mm⋅0.0125mm) ⋅ (4−pi)) / (4⋅0.0125mm) = 15.710646mm Ouch.
1. For perimeters and everything but internal infill, it's capped to either  the nozzle diameter * 1.05 or nozzle diameter *1.7
1. Top/bottom fill don't have the same role as internal infill, so they are capped like everything else to nozzle_diameter * 1.7.

This is either a bug or an unintended consequence of the heuristic used to calculate a 'sane' default, and the preference for wide traces. I think that we should cap the max at nozzle_diameter*3 as a general "good enough" solution. We still get the benefit of wider traces in areas we can't see it in, but it doesn't get absurd.

lordofhyphens, 2017-01-27:
[image]

Here it is limited to 3x nozzle_diameter.

bubnikv, 2017-01-27:
Did you ever try to print a 3x nozzle dmr extrusion? This is an insane
number.  the original intent I believe is to allow for a bridging flow for
infills that require it, like the rectilinear, or for infills with
incomplete overlap (honeycomb, cubic). But extruding extremely thin
extremely wide traces is physically impossible, if these traces are fully
supported. The plastic is simply too viscous for that. I know nobody, who
lets Slic3r use these "default sane" extrusion widths. Everybody defines
them explicitly.

forReason, 2017-01-27:
@bubnikv I guess many peoble use the simple mode because they don't want to mess with the lots and lots of advanced settings. Just had one requesting for Infill every X layers. Seems he uses Simple mode. In fact I also have not bothered with extrusion widths yet.

bubnikv, 2017-01-27:
I don't know your printer, but for the common Prusa i3 clones our settings
are probably a good start:
https://github.com/prusa3d/Slic3r-settings/tree/master/Slic3r%20settings%20175

forReason, 2017-01-27:
well. For me my settings are working really well. This may be since im using the latest stable Release 1.2.9 which came out 2015

lordofhyphens, 2017-01-27:
@bubnikv fair, although 3x nozzle diameter is certainly much better than
40x ;-) feel free to propose something different in the PR.

The current auto_width formula basically falls apart at small layer
heights, and the comment is the source is opaque to me still.

@forReason you want to try a print with that PR in place?

bubnikv, 2017-01-27:
Actually the maximum reasonable extrusion width decreases with the layer
height due to the viscosity of the extrudate. Thinner layer height means a
lot of pressure needs to be exercised to fill in that narrow space. Often
this is not physically feasible and it leads either to filament grinding,
or extruder skipping steps, or lifting of the print head to create a
necessary gap.

Yet worse, the nozzle does not usually have a flat tip of 3x diameter of
the orifice. So you will extrude and iron something over the lower layer,
which will hardly look flat, but more something like a rounded log with a
flute at the top.

I think the statement by @alexrj is as follows: As the infills are laid
using the bridging flow, extrude as much as possible to fill in the infill
region as quickly as possible.

I don't have any good proposal for the automatic extrusion width, but
please understand that this causes a major confusion to our users as the
"automatic" option is the default. My only proposal is if it does not work,
disable it.

forReason, 2017-01-27:
Unfortunately I can't print at the moment, as I scraped my good ol' Mendelmax 1.5 for a new, massive custom build 😞

PR #3691 (2017-03-11): Reduces the maximum flow for internal infill to 3x nozzle diameter
(avoid absurdly wide traces at small layer heights).

Fixes #3688
Fixes #2599
commit a42c1645 (2017-03-11): Reduces the maximum flow for internal infill to 3x nozzle diameter (avoid absurdly wide traces at small layer heights).

Fixes #3688
Fixes #2599
closed 2017-03-11 by alexrj