slic3r/Slic3r #3689: Cannot open slic3r from a standard mac os account

ITperson1, 2017-01-26:
### Version
_slic3r-osx-uni-1-2-9-stable_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### mac os Sierra + 10.12.2

### Behavior
* _When opening Slic3r from a standard mac os account the icon opens in the dock, but only for a split second.  Then the icon disapears, the program will not open.  The program does open when logged into an mac os admin account._
* _Need to be able to open slic3r from a standard mac os account running Sierra 10.12.2_
 

I logged into a admin account.  Then ran the dmg of slic3r 1.2.9 stable and dragged the icon to the applications folder.

lordofhyphens, 2017-01-26:
These are built on 10.6, and I've heard other reports of the builds not working on newer versions. I am in the (slow) process of getting a OSX automatic build going, negotiating with MacStadium for a CI server for the project.

lordofhyphens, 2017-01-26:
You may wish to peruse the instructions in the [wiki](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X) and build from source in the meantime. 

Once the CI server is going I'll see what can be done to support newer OSX (also surprised that OSX even has this problem).

alexrj, 2017-03-18:
@ITperson1, we now have automated builds, available at http://dl.slic3r.org/dev/
Can you confirm they work for you?
Thank you.

bubnikv, 2017-03-18:
Some users may have locked down accounts, which disallow execution of unsigned applications.

ITperson1, 2017-03-18:
Thank you.  I will try using this on Monday, will get back to guy and
report if it works.

lordofhyphens, 2017-03-30:
@ITperson1 It should probably work now, the new builds on OSX are signed (that change happened today). 

Please report back so we can close this issue.

lordofhyphens, 2017-04-04:
@ITperson1 Do the new builds work now?
http://dl.slic3r.org/dev/mac/

ITperson1, 2017-04-04:
Tested the new build, opened 1.3.0 dev in a standard account, slic3r opens
in a standard account now

closed 2017-04-04 by lordofhyphens