slic3r/Slic3r #3690: Which OSX version should we use for auto builds?

lordofhyphens, 2017-01-26:
We're going to be getting an OSX build/CI server shortly, thoughtfully sponsored by [MacStadium](http://macstadium.com). I will have the output on bintray, similar to the Windows system builds. 

We have a choice of OSX 10.7 through 10.12. As I understand it, that will provide a minimum OS for compatibility purposes.

Currently @alexrj develops on 10.6 and that's what the official builds are made for. We've seen some issues with those builds not working on 10.12. There were some bigger changes done in 10.7, etc.

I'm not an OSX person myself, so I'd like some feedback from the people paying attention around here that use Macs.

Any thoughts, @bubnikv?

bubnikv, 2017-01-27:
We are building on OSX 10.9. I have received a single report from a
customer, who has an issue running it. All others are fine with it. We
looked at the OSX versions logged into our web access log and the older OSX
versions are extremely rare.

I believe the OSX installs starting with 10.7 are free on Apple devices,
including virtualization running on an apple hardware, so we bought a Mac
mini and the 2parallels virtualization suite to build and test on. I have
provided all our build scripts to @alexrj.

lordofhyphens, 2017-01-27:
@bubnikv Yes, I saw that. @alexrj mentioned it when I had forwarded him the offer I had gotten from MacStadium. 

I know that 10.7 was the earliest blessed by Apple for virtualization. AFAIK we will have a bare-metal Mac Mini for our CI server, provided by MacStadium. If I make any modifications to the build scripts, I will of course provide them to you. I have been meaning to make a similar setup for Linux so that I can use the Travis CI server to do some deployment to BinTray. With the new local::lib, I think that PAR will be usable in the interim. Once the UI gets converted over to a native application and most of the Perl (and its dependencies) is gone I  think we can use cross-compilation instead of a native OSX server.

Thanks for the comment re: your download frequencies. I guess that if @alexrj wants to run 10.6 builds he can run the build scripts once a week or so. 10.9 is after the "hump" of changes that I remember hearing about as well.

bubnikv, 2017-01-27:
> @bubnikv <https://github.com/bubnikv> Yes, I saw that. @alexrj
<https://github.com/alexrj> mentioned it when I had forwarded him the offer
I had gotten from MacStadium.

Thanks @lordof, good to know @alexrj received my e-mail.

> With the new local::lib, I think that PAR will be usable in the interim.

For Linux, I think the appimage is probably the way to go for the binary
builds. A volunteer has provided a starting point for the appimage and it
seems to be working. Once ready for a next release, I will probably try to
add it to our build system.

https://github.com/prusa3d/Slic3r/issues/78

lordofhyphens, 2017-01-27:
https://en.wikipedia.org/wiki/AppImage Well, seems useful and not a sin against admin and man.

lordofhyphens, 2017-02-08:
10.9 it is.

closed 2017-02-08 by lordofhyphens

lordofhyphens, 2017-02-08:
[image]