slic3r/Slic3r #3692: Extrudate (gap fill) is too wide

koreywithak, 2017-01-27:
### Version
slic3r-1.3.0-dev.2016.1.1.2be6e1a
and older versions for at least a year

### Operating system type + version
Windows 10

### Behavior
When using default print settings and a layer height of 0.25 mm (in this case), the extrudate on a thin section is much too wide. This happens to me under many such scenarios. This can be avoided by manually setting extrusion widths with advanced settings. Turning off gap fill (setting gap fill speed to 0) will remove this extrudate, so I assume it is a gap fill issue.
Some more details from me and some pictures are found in this other post starting in January 2016: https://github.com/alexrj/Slic3r/issues/2599
It may be related to https://github.com/alexrj/Slic3r/issues/3688, but I'm not sure

#### STL/Config (.ZIP) where problem occurs
[Slic3r issue.zip](https://github.com/alexrj/Slic3r/files/734247/Slic3r.issue.zip)

lordofhyphens, 2017-01-27:
#3688 is a function of there being no maximum width and the automatic
formula gets bigger and bigger with smaller layer heights.

Gap fill logic is not the same as infill logic.

koreywithak, 2017-01-27:
Okay, thanks. From the linked issue where I started discussing this (https://github.com/alexrj/Slic3r/issues/2599), the measurement I gave of 2.4 mm was taken with calipers on the printed part with the poor extrusion. On a piece I printed with manual extrusion widths, it was very near 1.5 mm. I never noticed that the huge extrusion width in Slic3r had such a drastic effect on the final dimensions.

lordofhyphens, 2017-01-27:
@koreywithak Yeah, I measured the STL with netfabb basic to find out what the wall thickness is supposed to be. There were some nasty bugs with overextrusion of gapfill in 1.2.9 (mostly fixed now).

#3688 is specifically occurring at very low ( < 0.1mm) layer heights as well, just FYI. 

I don't think this is fixable without an overhaul of the perimeter generation (basically to reduce flow and try again if it can't fit the requested number of lines for those specific perimeters probably would be an okay heuristic). Right now Slic3r reduces lines instead of flow. 

To keep things fast, it is probably reasonable to switch to external perimeter flow instead of trying to step down (as at default external perimeters will always be thinner than regular perimeters)