slic3r/Slic3r #3693: Ya it crashes on my Mac as well, here is the crash output

Crisis64, 2017-01-28:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Crisis64, 2017-01-28:
Process:               slic3r [12713]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [12713]
User ID:               501

Date/Time:             2017-01-27 20:42:26.403 -0500
OS Version:            Mac OS X 10.12.2 (16C67)
Report Version:        12
Anonymous UUID:        ADE6577C-A3B8-3196-B41C-2606D65B6D8E

Sleep/Wake UUID:       24FEC5B5-E5AE-44E9-A394-92416F991DF7

Time Awake Since Boot: 42000 seconds
Time Since Wake:       96 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000000b
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0xb:
--> 
    __TEXT                 0000000000001000-0000000000038000 [  220K] r-x/rwx SM=COW  /Applications/Slic3r.app/Contents/MacOS/slic3r

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libperl.dylib                 	0x001fd1cc Perl_pp_entersub + 1564
1   libperl.dylib                 	0x0018abcd Perl_call_sv + 509
2   libperl.dylib                 	0x001dffcc S_invoke_exception_hook + 492
3   libperl.dylib                 	0x001e1e33 Perl_vwarner + 243
4   libperl.dylib                 	0x001e1eac Perl_warner + 44
5   libperl.dylib                 	0x00206a76 Perl_sv_free + 182
6   libperl.dylib                 	0x0023041f Perl_free_tmps + 111
7   libperl.dylib                 	0x0018e1c6 perl_destruct + 182
8   threads.bundle                	0x0485bfc4 S_ithread_clear + 196
9   threads.bundle                	0x0485f9ba XS_threads_detach + 458
10  libperl.dylib                 	0x001fd4a6 Perl_pp_entersub + 2294
11  libperl.dylib                 	0x001fbaf6 Perl_runops_standard + 22
12  libperl.dylib                 	0x0018aca5 Perl_call_sv + 725
13  Wx.bundle                     	0x0626fe67 wxPliEventCallback::Handler(wxEvent&) + 1111
14  libwx_baseu-3.0.0.0.0.dylib   	0x06d27d9f wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 95
15  libwx_baseu-3.0.0.0.0.dylib   	0x06e6f7a8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 104
16  libwx_baseu-3.0.0.0.0.dylib   	0x06e6ff98 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 88
17  libwx_baseu-3.0.0.0.0.dylib   	0x06e70904 wxEvtHandler::TryHereOnly(wxEvent&) + 52
18  libwx_baseu-3.0.0.0.0.dylib   	0x06e709ac wxEvtHandler::ProcessEventLocally(wxEvent&) + 60
19  libwx_baseu-3.0.0.0.0.dylib   	0x06e70a21 wxEvtHandler::ProcessEvent(wxEvent&) + 81
20  libwx_baseu-3.0.0.0.0.dylib   	0x06e71daf wxEvtHandler::ProcessPendingEvents() + 319
21  libwx_baseu-3.0.0.0.0.dylib   	0x06d28f20 wxAppConsoleBase::ProcessPendingEvents() + 144
22  libwx_baseu-3.0.0.0.0.dylib   	0x06e39abb wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 155
23  com.apple.CoreFoundation      	0x946d067e __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 30
24  com.apple.CoreFoundation      	0x946d05d7 __CFRunLoopDoObservers + 391
25  com.apple.CoreFoundation      	0x946b0972 __CFRunLoopRun + 1058
26  com.apple.CoreFoundation      	0x946b02ea CFRunLoopRunSpecific + 506
27  com.apple.CoreFoundation      	0x946b00db CFRunLoopRunInMode + 123
28  com.apple.HIToolbox           	0x93dafdb6 RunCurrentEventLoopInMode + 268
29  com.apple.HIToolbox           	0x93dafa77 ReceiveNextEventCommon + 195
30  com.apple.HIToolbox           	0x93daf99b _BlockUntilNextEventMatchingListInModeWithFilter + 83
31  com.apple.AppKit              	0x9257b641 _DPSNextEvent + 1227
32  com.apple.AppKit              	0x92cdf878 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1742
33  com.apple.AppKit              	0x92cdf1a2 -[NSApplication(NSEvent) nextEventMatchingMask:untilDate:inMode:dequeue:] + 132
[... 438 lines cut ...]
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x0000, -
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x0000, -
Memory Module: BANK 0/DIMM1, 4 GB, DDR3, 1333 MHz, 0x0000, -
Memory Module: BANK 1/DIMM1, 4 GB, DDR3, 1333 MHz, 0x0000, -
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x9A), Atheros 9380: 4.0.74.0-P2P
Bluetooth: Version 5.0.2f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: ST31000528AS, 1 TB
Serial ATA Device: HL-DT-STDVDRW  GA32N
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: USB2.0 Hub
USB Device: Multi-Reader
USB Device: USB2.0 Hub
USB Device: FT232R USB UART
USB Device: IR Receiver
USB Device: Card Reader
Thunderbolt Bus: iMac, Apple Inc., 25.1

lordofhyphens, 2017-01-28:
Need you to describe what you were doing (reproduction steps). Those generic crash logs aren't very useful on their own.

Crisis64, 2017-01-28:
I had an stl file open, & was just adjusting speed & brim distance, & it crashed

lordofhyphens, 2017-01-28:
Turn off background processing.

Crisis64, 2017-01-28:
Ok I have turned off background processing, will let you know

Astroman, 2017-01-29:
My crashes have been pretty constant lately with no discernible trigger for them. Turning off background processing didn't help. Running Slic3r on a new touchbar MacBookPro OS X 10.12.3

alexrj, 2017-03-13:
Can you all try the fresh builds available at
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view ?

I expect them to be pretty stable. I'll leave this issue open until confirmation. Thank you!

alexrj, 2017-04-04:
@Astroman, @Crisis64, can you try a fresh build from http://dl.slic3r.org/dev/ and confirm this is fixed? Thank you!

Astroman, 2017-04-04:
Not sure which version you want me to try. Also, whichever ones I try to download, it returns “Forbidden!” error.

lordofhyphens, 2017-04-04:
Yeah, bintray is screwing with us at present. You can grab the newest from
http://dl.slic3r.org/dev/mac (that will likely be the -dev location going
forward).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Astroman, 2017-04-04:
I loaded and played with a fairly complex part that probably would have crashed before. No problems! Since it looks like you’ve added some nice features since I last used it, I’ll probably come back to your version for my next project. Thanks!

alexrj, 2017-04-05:
Thank you very much for testing!

closed 2017-04-05 by alexrj