slic3r/Slic3r #3695: Slic3r version MacOS pixelation in iMac 5K

litris, 2017-01-31:
### Version
1.2.9

### Operating system type + version
macOS Sierra 10.12.3
iMac 27" 5K Late 2015 m395

### Behavior
In an imac 5K Slic3r you see a very low resolution it seems an erroneous scaling you see pixelated text without resolution.

Attached is a screenshot in window mode if wider is pixela more.

any solution?

In the capture of the window where it shows the version looks perfect but the slic3r window looks pixelated.

Thank you.
[image]
[image]

bubnikv, 2017-01-31:
See https://github.com/prusa3d/Slic3r/issues/97 for resolution.

litris, 2017-01-31:
It does not work for me :-(

I pasted the same text as you show and the text is pixelated in an imac 27 "5k late 2015.

alexrj, 2017-04-04:
This should be fixed as of now. Could you get a fresh automated build at http://dl.slic3r.org/dev/mac/ and confirm it works?  (Just the text, we still have the 16x16 icons)

lordofhyphens, 2017-04-11:
@litris can you check one of our automated builds to see if this works for you?

lordofhyphens, 2017-04-14:
We can probably use my hacked wxperl and try 3.1.0 on OSX.

litris, 2017-04-30:
Sorry for not answering sooner.
Yes! Works perfectly text and menus flout icons remain pixelated.

Thank you.

closed 2017-05-01 by alexrj