slic3r/Slic3r #3696: standalone exe won't slice amf on test system without perl installed

chrisatsea, 2017-01-31:
### Version
1.3.0-dev / 1.2.9-622-gd41e3c1
### OS
Win 10
### Dev System
Strawberry v 5.24.1 32bit, Boost 1.59.0
### Behavior
Slic3r build went flawless, "perl slic3r.pl" works as expected with amf and stl input file types. Packing an exe with 
''pp -M Encode::Locale -M Moo -M Thread::Semaphore -M Slic3r::XS -M Unicode::Normalize -M XML::SAX -M XML::SAX::ExpatXS -o slic3r.exe slic3r.pl''
creates a working slic3r.exe on development system that happilly takes amf as input and does it's job as expected, however, on test systems with no perl installed it throws this error while slicing amf files:

''Can't load 'C:\Users\USER~1\AppData\Local\Temp\par-43687269737469616e2056656974\cache-f77007bfcd81d35fec8b7e6afa9c1ce98622bb7c\a88e4e13.xs.dll' for module XML::SAX::ExpatXS: load_file:The specified module could not be found at C:/Strawberry/perl/lib/DynaLoader.pm line 193.
  at C:/Strawberry/perl/vendor/lib/PAR/Heavy.pm line 123.
Compilation failed in require at (eval 162) line 1.''

The dll is generated while executing slic3r.exe, so it's there....
stl-slicing works
...tried numerous packing solutions...none worked so far.
Without ExpatXS it throws "AMF parsing requires XML::SAX"...
...different amf files...freshly generated by Repetier-Host...nope...
...banging my head on the table didn't seem to work either...

Any suggestions?

Regards,
chris

lordofhyphens, 2017-01-31:
You will probably need to build with SLIC3R_STATIC defined.

Also, look at utils/package_win32.ps1

I would also remove Expat from your strawberry perl system, rebuild, and
include with -a the entire local-lib directory. Odds are -M didn't pull in
a few DLLs because they weren't exercised during the packing process.

chrisatsea, 2017-01-31:
Thanks for the quick answer!
Will try and tell.

chrisatsea, 2017-01-31:
Unfortunately, this hasn't worked so far. Removing the Expat- and two tries later the ExpatXS files from the system only led to attached error (not unknown to me, as it has been the reason to deal with ExpatXS in the first place...i guess...had so many building attempts today i most likely forgot some of the console outputs...)
SLIC3R_STATIC perl Build.pl wasnt recognized...(neither with ..=1)
Including the local-lib dir also didnt work. Could it be a problem with the SAX Module (or ExpatXS) Module itself? 
I am somehow desperate and thinking about to -l every dll i find in the perl-dir next time...^^
But first I will install a fresh copy of perl (again) now, rebuild everything (again)...

[error.txt](https://github.com/alexrj/Slic3r/files/742955/error.txt)

lordofhyphens, 2017-01-31:
I suggest using powershell and setting an environment variable (google for
details).

For the attached error you would need to make sure expat gets installed in
the local-lib dir, not in your base perl install.

I am certain that your underlying issue is that pp doesn't know to pack
that dll. Another solution would be to include the dll in the packed
archive. package_win32.ps1 does this for a couple gcc dlls.

You would probably have the same problem with my packed builds as well
(feel free to test), if I am right.

Frankly it would be easier (for me) to write a short c program that does
the same thing as the batch file to deal with RepetierHost  (which we don't
support and I don't use).

You would need to forward all of the arguments, of course, passed to it.

chrisatsea, 2017-01-31:
Well, it wasn't that hard afterwards: You were quite right with pp. 
Installed XML::SAX into slic3r's source \local-bin\...\ folder, made sure all the dlls where in place, and pp -a it togehter with -M XML::SAX. I did let it execute an amf while packing (although i don't think that was necessary).
Seems to work nicely on the naked test system with stl, amf and obj files. Size is about 12MB - no gui though.

Thank you for input and inspiration.
Chris

closed 2017-01-31 by chrisatsea

lordofhyphens, 2017-01-31:
I would purge your local perl install and reinstall it, just to make sure you have a portable version of the dependencies. 

At some point I will get a minimum bootstrap perl environment together so people can download a dependencies file (I already have one online on bintray.com for the build server's use). It is painful though

lordofhyphens, 2017-01-31:
Could you add a small section on the wiki page for Windows builds with your
experience?