slic3r/Slic3r #3698: Post proessing scripst not working?

Tinchus2009, 2017-02-07:
### Version
1.2.9-622-gd41e3c1

### Operating system type + version
win10

### Behavior
I have used some post processing scrips in the past, I tried to used one now, the weight cost script included in the utils folder. For some reason I can see this:

1.- 2 backslases are being added automatically in the script path, i get this for example: "F:\\Slic3r\\utils\\post-processing\\filament-weight.pl"

2.- .pl files are asociated to wperl. The slice finishes aparrently OK, but in the generated gcode there is no weight/cost information added.

In the post processing section I have this: "F:\\Slic3r\\utils\\post-processing\\filament-weight.pl"

removeing the 2 slashes doesn't work, they are added automatically as soon as I click in any other place of the screen.
Doing the slice through command prompt also fails in creating the extra info.

No errors reported. I have used this script in the past (I have to say the last time I used it was in the stale 1.2.9 version) with no problems

bubnikv, 2017-02-08:
Yes, it seems to be broken. It may be due to my addition of the filament notes and the required changes in the ini file parser.

bubnikv, 2017-02-08:
This patches the issue:
https://github.com/prusa3d/Slic3r/commit/6b1a72aac97d9cf87907eb3b5fd38b78a2859b50

Still I consider this a patch only. The current way of entering a script invocation does not allow a semi-colon inside the invocation line.

Tinchus2009, 2017-02-08:
sorry for the question, but how would be the proper way to apply the patch?

bubnikv, 2017-02-08:
If you pull the development code from the @alexrj repository using git,
you may try your luck by entering the Slic3r directory and running

git cherry-pick 6b1a72aac97d9cf87907eb3b5fd38b78a2859b50

which will apply the patch to your code repository.

Or you may pull the prusa3d fork and compile that.

bubnikv, 2017-02-08:
I forgot, it is a bit more complex than that with the git cherry-pick. You would have to add remote first:

git remote add prusa3d git://github.com/prusa3d/Slic3r

then to fetch that repository

git fetch prusa3d

then you could apply the patch from the prusa3d fork:

git cherry-pick 6b1a72a

lordofhyphens, 2017-02-08:
He needs to have your master fetched as well as a different branch to be
able to cherry pick.

The simpler way for him is probably to read the commit and apply it
manually  (lines starting with - mean it is removed, + means add).

More involved would be to clone your fork, make a patch out of the commit
and apply it.

Tinchus2009, 2017-02-08:
This is what I did. I manually changed the file. The patch seems to not 
work, scripts are still not working (I already built again before trying).

alexrj, 2017-03-01:
@bubnikv, are those *escape_string*_cstyle functions actually needed? Why can't we just replace newlines with `\n` and semicolons with `\;` ?

bubnikv, 2017-03-01:
It is for the filament comments.

alexrj, 2017-03-05:
Yes, why can't we replace `;` with `\;` and then use unescaped `;` for joining the strings?

bubnikv, 2017-03-05:
There are many ways to skin a cat.

alexrj, 2017-03-05:
@Tinchus2009, this is now fixed in the master branch. Thank you for the report.

closed 2017-03-05 by alexrj

Tinchus2009, 2017-03-06:
Hi. I have just builted the latest dev: 1.2.9-677-gf4d73f1

I did a test trying to use a script, but still it is not working, at least not under windows 10.
The script tested is  the one distributed: filament-weight

It has been modified to include the .bak command as instructed in the wiki: 
use constant CURRENCY => "EUR";

$^I = '.bak';

while (<>) {

When exporting the gcode, the model is sliced, the a command prompt opens and closes, and the gcode is exported, but inside the gcode there is no info about the filament weight/cost as it should be.
The perl scripts are associated in my system to be executed with wperl.exe. 
I have tried useing perl.exe but useing that makes the command prompts to open and never closes.
Te script is called inside slic3r useing the full path (in my case it is F:\Slic3r\utils\post-processing\filament-weight.pl)