slic3r/Slic3r #3700: Just an idea - automatic generation of "mouse ear" brim

DjDemonD, 2017-02-08:
_Is this a new feature request?_Y

I know many people printing ABS onto certain surfaces struggle with bed adhesion and using a brim is not always sufficient. Some advocate, and I use this a lot, using circular pads at the extremities of an object to improve adhesion "mouse ears" and these work extremely well. I've attached an image, one part in ABS with a 5mm brim with warp that has detached from the bed, and the same part with "ears" which has not.

Would it be possible to generate this type of support automatically? This would improve ABS (and other materials) adhesion and save users a fair bit of time adding these features, part by part in CAD? I suspect with them a brim is unnecessary and they may even be a more efficient use of filament, certainly easier to trim away afterwards.
[image]

konvoj, 2017-02-14:
I vote for this feature as well. ABS is nasty around the corners and standard brim generated by Slic3r is not enough for some prints.

alexrj, 2017-02-28:
#1316

lordofhyphens, 2017-03-10:
Did some initial investigation, I think collecting the convex points in the bottom islands would give you the correct origin points; center circles there and subtract the object (similar to how brims appear to be done).

lordofhyphens, 2019-01-11:
Fixed via #4691

closed 2019-01-11 by lordofhyphens