slic3r/Slic3r #3701: [Ask for information] "Can't locate object method" when adding a class with XS++

corthiclem, 2017-02-09:
Hi everyone,

I am trying to add a new feature in Slic3r by adding a C++ object. Basically, as an exercise, I tried to create a copy of the class OozePrevention in the file GCode.hpp (/.cpp) by simply changing the name.

I am quite new with XS and XS++ but I added conscientiously my class in the files typemap.xspt, my.map, perlglue.cpp, GCode.xsp and XS.pm as well as "OozePrevention" was added.

Slic3r built successfully. However, when I try to call one of the function created in the GCode.xsp file, I got this error: 

`Can't locate object method "set_enable" via package "Slic3r::GCode::OozePreventionCopy::Ref at C:/.../GCode.pm ... `

Do you have any clue where it comes from? I know this is not a issue that affect Slic3r usage, but I would be pleased to understand how to make such changes in Slic3r.

Thank you in advance!

### Version
I am using the dev version and the last commit. 
I use Perl 5.18

### Operating system type + version
I use Windows 7

lordofhyphens, 2017-02-09:
Look in xs/xsp directory if you are trying to make it accessible from Perl.

corthiclem, 2017-02-10:
@lordofhyphens 
I added the new functions and the new class in the xsp by adding:

`
%name{Slic3r::GCode::MyClass} class MyClass{

    MyClass();
    bool enable()
        %code{% RETVAL = THIS->enable; %};
    void set_enable(bool value)
        %code{% THIS->enable = value; %};
};
`

Anyways, reading #2276, I restarted from scratch and compiled a fresh Slic3r. It did the trick! ;)

Thanks for your answer.

lordofhyphens, 2017-02-10:
Yeah, whenever you mess with the xsps it's wise do to a full rebuild. 

Is there a Feature Request documented in the issue tracker for what you are planning on doing? If you are going to submit a PR I'd like to have an existing issue to link to, preferably with a discussion of what is desired. 

It's also helpful to write some tests for your new feature too. (very very helpful).

closed 2017-02-10 by lordofhyphens