slic3r/Slic3r #3702: class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]

xoan, 2017-02-09:
It is only a warning when compiling (and it is not new) in currently master:

```
src/libslic3r/Print.cpp:886:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<Fill> filler(Fill::new_from_type(ipRectilinear));

src/libslic3r/SLAPrint.cpp:60:14: aviso: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<Fill> fill(Fill::new_from_type(this->config.fill_pattern.value));

src/libslic3r/SLAPrint.cpp:187:14: aviso: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<Fill> fill(_fill->clone());
```

Seems to be fixed by changing `auto_ptr` to `unique_ptr`.

lordofhyphens, 2017-02-09:
That's down to your compiler, older compilers wouldn't show the warning (and the older compilers won't have `unique_ptr`. 

@alexrj I'd very much prefer to require at a minimum a C++11-compilant compiler at this point.

So this would be clang 3.3 or higher, gcc 4.8 or higher, or MSVC 14.0

http://en.cppreference.com/w/cpp/compiler_support

xoan, 2017-02-09:
My bad, sorry. The report is very much incomplete :\

### System
```
% uname -a
Linux 4.4.47-1-lts #1 SMP Sat Feb 4 18:19:37 CET 2017 x86_64 GNU/Linux

% g++ --version
g++ (GCC) 6.3.1 20170109
```

lordofhyphens, 2017-02-09:
Yeah that is a fairly new compiler ;-)

PR #3775 (2017-03-19): remove deprecated code and add new one
change auto_ptr to unique_ptr

Fixes : #3702
closed 2017-03-19 by lordofhyphens