slic3r/Slic3r #3704: [Feature request] Overhang-sensitive Z-lift

VanessaE, 2017-02-11:
After some conversation in IRC while trying to help Artesian3D sort out a problem with a print, it occurred to me that a feature is needed in Slic3r where it will induce Z-lift if the nozzle would pass over any part of a bridge or overhang during a travel move.  That is, all normal travel moves would be done without Z-lift, if it isn't enabled.  The idea is to avoid hitting curled-up ends of overhangs such as the top portion of a horizontal hole, or the ends of a bridge.  

In such a case, the nozzle should perhaps raise and lower smoothly (if quickly) as it approaches and leaves the "danger zone", retracting quickly at the start of the travel (but without interrupting it), and perhaps starting and ending the lift/drop a millimeter or two from each end of the travel move, so that there are no stop-lift-go artifacts.

Artesian3D, 2017-02-11:
Desperately needed. Good idea.

alexrj, 2017-05-01:
Why not just using the "Avoid crossing perimeters" feature? That's its purpose.

VanessaE, 2017-05-01:
"Avoid crossing" wouldn't be able to account for the vertical distortion that always happens with an overhang - that is, it wouldn't be able to avoid crossing a perimeter that started the overhang a few layers below the current layer, it would only be able to take into account the position of the associated perimeter on the current layer.

Besides, I've seen comments that the "Avoid crossing" feature can really slow down a print, between slicing time and non-optimal travel paths.