slic3r/Slic3r #3705: [feature request] "External perimeters first" automatic mode

peternann, 2017-02-11:
It seems that "external perimeters first" is more likely to get good results with models (/layers?) that don't have over-hanging angles. Occasional (if not regular) small blobs of excess extrudate at the outer layer are a common feature on many of my prints, often due to imperfect bed levelling I think. (Perfect levelling is hard!)

But the default in Slic3r is to have "External Perimeters first" turned off, just in case there are over-hangs I suppose (Which, I agree, is quite common).

Couldn't Slic3r decide if there is an over-hang or not, and hence decide whether 'internal first' is really needed or not?
If not as a 'global decision', perhaps it could decide layer-by-layer and perimeter-by-perimeter?
How? I've got little idea how the code works, but one algorithm thought is:
Compared to the same perimeter outline for the previous level:
1) Does this outer perimeter ever cross the previous one in X-Y space?
2) Is the total perimeter distance longer that the last one

If yes to either one, you have an overhang.
For internal holes, reverse the tense of question (2).

Other algorithms may use the orientation of relevant surface polygons?

I'm only guessing.

Probably tremendously harder than it sounds, I'm sure.

Just a suggestion. Would seem to improve the quality of my prints.

alexrj, 2017-03-18:
That option was introduced in order to help with dimensional accuracy, but I expect quality to be lower anyway. Changing it on a per-layer basis would make quality worse because layers would look different...

peternann, 2017-03-19:
I get noticeably better prints with 'external perimeters first'.

But as I said - my printer (and skills) aren't great! Maybe I have an extrusion calibration problem and I'm regularly over-extruding.

But I see your point - Changing it layer-by-layer would be bad. Scratch that thought.

My plan at this point is to leave 'external perimeters first' unless I'm printing with an overhang. Seems to benefit my prints.

Cheers.

closed 2017-03-19 by lordofhyphens