slic3r/Slic3r #3707: Orbit center not updated when scaling model

hexane360, 2017-02-12:
### Version
1.2.9-620-g1d02c15

### Operating system type + version
Ubuntu 16.04

### Behavior

Sometimes I download models that are comically large, and scale them down to something like 2% in Slic3r. However, when I do this, the orbital center stays in the same Z position (1/2 the model's size). This makes orbiting around the model very difficult.

The solution to this should be to have a scaling operation update the Z position of the orbit center.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
N/A

uclaros, 2017-02-18:
Since pull request #3556, you can use "alt+right click" to move the center of orbit up or down. You can also "double-middle-click" to reset the view and set the center of orbit to the center of all objects on the plater. I don't know if this is documented yet.
Automatically resetting the orbit center when scaling seems logical when having a single object on the platter, however if you have more than one objects you might not want the center of orbit to reset when you scale one of them.
Maybe the reset view should be added to the Plater menu?

hexane360, 2017-02-25:
Alt+right click works great for me (well, after I stopped ubuntu squashing the shortcut--not your fault). However, I don't believe its in documentation anywhere. That on its own might be a good compromise. Maybe it could go in the ["working with models"](http://manual.slic3r.org/first-print/working-with-models) section of the manual. That whole page seems woefully outdated. If I find time I'll work on a PR to update it.

closed 2017-02-25 by hexane360