slic3r/Slic3r #3709: Thin walls path turn at the end

xoan, 2017-02-15:
### Version
current master: 402035732b2f3e0bbcae98d740ef97be8d2edbfd

### Operating system type + version
Arch Linux: 4.4.48-1-lts x86_64

### Behavior
In some thin walls, the toolpath makes a strange turn at end, but not in all layers.

[image]

#### STL/Config (.ZIP) where problem occurs
* [coaster_holder.stl](http://www.thingiverse.com/thing:154847/#files)
* [FoldaRap - 1.3.0-dev.ini](https://gist.github.com/xoan/ea2768b728b91e4624cdbf575030b3bd)

bubnikv, 2017-02-16:
The thin walls are based on a skeleton of the slice:
http://homepages.inf.ed.ac.uk/rbf/HIPR2/skeleton.htm

What you are seeing is following:
[image: Inline image 1]

[image: Inline image 3]

[image: Inline image 4]

Depending on your luck, different end segments are taken. This is not ideal
and it has to be improved. Namely, something like this shall be extruded:

[image: Inline image 5]

bubnikv, 2017-02-16:
The thin walls are based on a skeleton of the slice:
http://homepages.inf.ed.ac.uk/rbf/HIPR2/skeleton.htm

What you are seeing is following:
Inline image 1
[image]

Inline image 3
[image]

Inline image 4
[image]

Depending on your luck, different end segments are taken. This is not ideal and it has to be improved. Namely, something like this shall be extruded:

Inline image 5
[image]

alexrj, 2017-03-18:
There's a pruning algorithm which does exactly what described in the last picture. It's based on heuristics and it's not always easy to define a behavior, but it can be definitely improved.