slic3r/Slic3r #3710: Error when building OpenGL-0.6704

HaoTang1, 2017-02-16:
### Version
1.29.0

### Operating system type + version
Windows 10 64bit and Windows 8 64bit
Strawberry Perl 
VS 2013 12.0.40629.00 Update 5

### Behavior
Error when building OpenGL using nmake

I've followed the [Wiki](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows) and comments on [#3555](https://github.com/alexrj/Slic3r/issues/3555),

`perl Makefile.PL verbose interface=FREEGLUT`   this one worked and the MakeFile got generated correctly.

Then `nmake` failed with the following error.

c:\OpenGL04>nmake

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Skip blib\lib/OpenGL/Array.pod (unchanged)
Skip blib\lib/OpenGL.pod (unchanged)
Skip blib\lib/OpenGL.pm (unchanged)
cp Config.pm blib\lib/OpenGL/Config.pm
Skip blib\lib/OpenGL/Tessellation.pod (unchanged)
Running Mkbootstrap for OpenGL ()
        "C:\wperl64d\bin\perl.exe" -MExtUtils::Command -e chmod -- 644 "OpenGL.bs"
        cl -c    -nologo -GF -W3 -Od -MD -Zi -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY /ogity /GDs    -DVERSION=\"0.6704_93\"  -DXS_VERSION=\"0.6704_93\"  "-IC:\wperl64d\lib\CORE"  -DHAVE_VER -DGL_VERSION_USED=4.5 -DHAVE_FREEGLUT -DHAVE_FREEGLUT32 -DHAVE_GL -DHAVE_GLU -DHAVE_GLU32 -DHAVE_OPENGL32 OpenGL.c
cl : Command line warning D9035 : option 'o' has been deprecated and will be removed in a future release
cl : Command line warning D9002 : ignoring unknown option '/GD'
OpenGL.c
        cl -c    -nologo -GF -W3 -Od -MD -Zi -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY /ogity /GDs    -DVERSION=\"0.6704_93\"  -DXS_VERSION=\"0.6704_93\"  "-IC:\wperl64d\lib\CORE"  -DHAVE_VER -DGL_VERSION_USED=4.5 -DHAVE_FREEGLUT -DHAVE_FREEGLUT32 -DHAVE_GL -DHAVE_GLU -DHAVE_GLU32 -DHAVE_OPENGL32 gl_util.c
cl : Command line warning D9035 : option 'o' has been deprecated and will be removed in a future release
cl : Command line warning D9002 : ignoring unknown option '/GD'
gl_util.c
gl_util.c(956) : warning C4244: 'initializing' : conversion from 'IV' to 'unsigned long', possible loss of data
gl_util.c(965) : warning C4244: 'initializing' : conversion from 'IV' to 'unsigned long', possible loss of data
gl_util.c(976) : warning C4244: 'initializing' : conversion from 'IV' to 'unsigned long', possible loss of data
gl_util.c(984) : error C2036: 'void *' : unknown size
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\cl.EXE"' : return code '0x2'
Stop.

I've tried OpenGL-0.6704_07 and OpenGL-0.6704_93, but got the same error.

Thanks!

lordofhyphens, 2017-02-16:
0.70 should work just fine unless you're using an old version of Slic3r. That would be my next suggestion.

OpenGL.pm isn't our code, so problems compiling isn't something we can help with directly (I don't use VS). 
@bubnikv apparently builds on VS, though. 

For Strawberry perl/mingw: I've made an archive for all the dependencies for 32-bit Strawberry perl 5.24.0 (compiling with mingw). https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.24.0-win32.7z

lordofhyphens, 2017-02-22:
64-bit archive https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev

There's really nothing else we can do about this, so closing.

closed 2017-02-22 by lordofhyphens

HaoTang1, 2017-02-22:
Thank you so much for your help.

gshasdy, 2017-03-03:
Having the same issue. @lordofhyphens  Do you by any chance have a 64-bit archive for Strawberry perl 5.24.0 instead of 5.18.0? Thanks.

gshasdy, 2017-03-03:
Deleted my comment for not reading the instruction carefully. Thanks.

lordofhyphens, 2017-03-03:
@gshasdy as a FYI, there's some known issues with a hanging bug with 64bit perl past 5.18 (it's a perl/strawberry perl thing), which is why the 64bit build is for 5.18. 

The bug doesn't exist for 32-bit,; which is why that dependency file is 5.24

bubnikv, 2017-03-03:
I plan to put together a binary perl distribution for both 32bit and 64bit
windows and Visual Studio 2013. That will be free of this croak / C++
exception issue and I also expect that the Visual Studio compiler will
generate a better code than some outdated GCC packed with Strawberry Perl.