slic3r/Slic3r #3712: The program 'slic3r.pl' received an X Window System error.

patrickelectric, 2017-02-16:
IDK the version, I can't start the program. 
Installed via arch aur git package.
```
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 983 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

lordofhyphens, 2017-02-17:
Since I can't reproduce on Debian, you should probably contact the package maintainer first.

lordofhyphens, 2017-02-17:
I at this point can only suggest installing perl local-lib from aur packages and building and running Slic3r from source using a checkout of this upstream source and running `perl Build.pl`.

lordofhyphens, 2017-02-17:
Do not build an older version of Slic3r than commits from January or you will wreck your distro's Perl install. 

DO NOT BUILD WITH ROOT/SUDO.

lordofhyphens, 2017-02-17:
In order to have a clue as to what is going on, we're going to need to know what you did to build. I don't use Arch, and I'm not setting up an Arch distro.

swiftgeek, 2017-02-17:
7d2884bc0c3a492e2164f6d1950c1391ab872bcd on master branch works fine for archlinux, you probably compiled stable branch

alexrj, 2017-05-16:
I think we can close this as we couldn't reproduce it and it's clearly an upstream issue of wxWidgets and/or GTK.  Things will definitely improve on our side when we finish the C++ port of the GUI.

closed 2017-05-16 by alexrj