slic3r/Slic3r #3714: Building just modified CPP/HPP + dependencies (for linking)...

jaggzh, 2017-02-17:
In Debian Jessie, I've been trying to add some features, but the Build.pl recompiles all (or a lot of) the xs files each time, even with one small change, and I think even with no modifications.  I'm not familiar with using perl to build projects, and wanted to confirm whether or not it actually examines dependencies of the .cpp and .hpp files -- it seems it doesn't?

Thanks a bunch.

bubnikv, 2017-02-17:
You may take a shortcut by calling just
perl xs/Build

but it does not work reliably. Namely, if you change a header, it will not recompile the cpp files including it.

Heck, the real programmer does not complain that the compilation process takes a long time. He uses the opportunity to take a nap.

lordofhyphens, 2017-02-17:
If you jump into the xs folder, there is another Build.pl file and a
supplementary script that it generates  (Build) that does the work.

I don't remember if it obeys local::lib when you do that and it isn't very
smart

closed 2017-02-17 by jaggzh

lordofhyphens, 2017-02-17:
Just as a FYI, there be dragons lurking there. If you get weirdness, drop
back to the main source dir and rebuild everything.

http://search.cpan.org/~leont/Module-Build-0.4220/lib/Module/Build.pm <--
that's the Perl module that implements the build system.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.