slic3r/Slic3r #3715: Raft expansion

jaggzh, 2017-02-17:
Depending on a model's structure, a raft may be divided into islands -- sometimes too small for some adhesion issues.  Being able to increase a raft's extension beyond the object/support on top of it would be useful.  I noticed a setting, raft_offset, which wasn't exposed in the UI, and modified the code to expose it.  However, in my tests it doesn't seem to do what I'm looking for (actually it seems to have no effect at all, and I'm not yet sure what it's actually controlling).

At xs/src/libslic3r/SLAPrint.cpp:138, I even hardcoded tests, like:
-        raft = offset_ex(raft, scale_(this->config.raft_offset));
+        //raft = offset_ex(raft, scale_(this->config.raft_offset));
+        raft = offset_ex(raft, scale_(100));

Still nothing.

Here's an example model where some detached "islands" ended up breaking off:
[image]

Having the option to expand the raft by some mm's would be useful in this case, I think, but I'm not yet sure how to do it.  Eventually that offset_ex() calls _offset(const Polygons &polygons, const float delta, double scale, ClipperLib::JoinType joinType, double miterLimit)...
...but I'm not sure what that function's actually doing either.

lordofhyphens, 2017-02-17:
For a raft, I agree that projecting the model down and using that to generate the raft is probably the safest approach.

jaggzh, 2017-02-17:
Projection method might be a nice option to have, but I think it might be
an uncommon need (usually inefficient)?  Are there other complex situations
where it would help? Minimum raft island area might be a solution, while
I'm hoping the option to just expand the raft is the most simple to code in
(being brand new to this).

<https://github.com/alexrj/Slic3r/issues/3715#issuecomment-280731868>, or mute
the thread

bubnikv, 2017-02-18:
Did you test the new supports in the prusa3d fork? It expands the 1st layer
around the raft and the support columns.

jaggzh, 2017-03-07:
bubnikv, no, only the main fork.  I'll check out the prusa3d to see if it has the changes I need.  Thanks.

jaggzh, 2017-04-04:
Here's a copy of the .stl. New builds' supports climb all over my objects so I'm unable to reproduce this raft issue for the time being.

[body.zip](https://github.com/alexrj/Slic3r/files/894319/body.zip)