slic3r/Slic3r #3719: Citrus Perl 5.14

diegoolavarria, 2017-02-20:
Is there a link where I can download CitrusPerl 5.14? in their website there's only version 5.16, which it can`t be used with Slic3r.

lordofhyphens, 2017-02-20:
Not that I am aware. I suspect that their own site has a link somewhere. 

I use Strawberry Perl myself, and that's what the build server output is used with:
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view

closed 2017-02-20 by lordofhyphens

diegoolavarria, 2017-02-20:
I tried with Strawberry, but then I couldn't create an executable. I used Cava Packager, which only works with CitrusPerl; Perl2exe, whick just couldn't even; and Par::Packer, which after a lot of trying just doesn't works with Strawberry.

After reading that Cava Packager works with Citrus, I installed the 5.16 Citrus version, but it had bring me nothing but errors and missing modules :/

bubnikv, 2017-02-20:
@diegoolavarria Don't use Cava Packager. AFAIK it bundles very old wxPerl, resulting in unstable Slic3r.
The Prusa3D builds are done with Strawberry Perl as well.

diegoolavarria, 2017-02-20:
@bubnikv you mean the 5.18 version of Strawberry?

lordofhyphens, 2017-02-20:
Yes, 5.18 for x64 because of other perl issues.

Why must you have a packaged exe binary?

I know, I am trying to de-Perl as fast as I can.

lordofhyphens, 2017-02-20:
Sigh, I need to write a short windows program that all it does is call
Slic3r as a subprocess and pipe its stdio.

diegoolavarria, 2017-02-20:
@lordofhyphens We are trying to distribute our own version of Slic3r, like Prusa did.

lordofhyphens, 2017-02-20:
Ah. Why?

diegoolavarria, 2017-02-20:
@lordofhyphens Because we are from another country, so I had to translated it, and create new custom preferences to be used with our printers.

bubnikv, 2017-02-20:
@lordofhyphens
> Sigh, I need to write a short windows program that all it does is call Slic3r
as a subprocess and pipe its stdio.

What do you mean?
I have a slic3r.exe, a short wrapper, which loads and executes perl.dll. I
think you should have it, I gave it to @alexrj.

bubnikv, 2017-02-20:
> I know, I am trying to de-Perl as fast as I can.

By the way, our builds are very stable, I don't see an immediate need for
de-perl. IMHO there are much more important stuff to do.

lordofhyphens, 2017-02-20:
I don't think it got over to me, just the OSX link (which I don't remember if my access to that google drive works).

#3720 should behave itself, a basic stubbed out "exe" that shells out to Perl.

AFAIK our builds are also stable, de-perling mostly removes those dependencies and quite a bit of the headaches (for me, dealing with other people).

@diegoolavarria I'm actually working on i18n support for Slic3r right now.