slic3r/Slic3r #3721: "Avoid Crossing Perimeters" is crossing internal small perimeters (holes)

cobra18t, 2017-02-21:
### Version
1.2.9

### Operating system type + version
Win7 SP1 64bit

### Behavior
* With "Avoid Crossing Perimeters" enabled, there are many occasions where a travel move within an object crosses over an internal small perimeter. 

* _Steps needed to reproduce the problem_
Slice Test.stl file with config.ini from the attached zip.
  
* _Expected Results_
I would expect the move to skirt around the already printed internal small perimeters
* _Actual Results_
Instead, the moves go right through the perimeters. Here I show two example layers. See green moves crossing over the 3mm holes. There was room between the holes to skirt around them. There is no retraction since it is above the "Minimum travel after retraction" distance. The causes a fair bit of extra material in the printed holes.
  [image]
[image]

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
[Slic3rIssue.zip](https://github.com/alexrj/Slic3r/files/789002/Slic3rIssue.zip)

7eggert, 2017-03-16:
I was just about to post a similar problem, except my print includes additional moves to the outside in order to carefully knock off everything:

[image]
[Unterteil.zip](https://github.com/alexrj/Slic3r/files/848875/Unterteil.zip)

Slic3r 1.2.9/Linux Debian Jessie

alexrj, 2017-05-16:
Hi @cobra18t and @7eggert, could you test this with 1.3.0-dev? It should be fixed. Get it at http://dl.slic3r.org/dev/

cobra18t, 2017-05-17:
It still seems like it is violating the inside small perimeters at times in 1.3.0-dev. You can see the travel move skirts around the hole in the upper right, but blows right through the lower hole:

[image]

7eggert, 2017-05-19:
Also it's still crossing already-printed support material. I only verified in http://gcode.ws/ yet.

Testcase: Attachement

Since I added the USB connectrion to the printer settings (possibly unrelated), the dev version of slicer causes an out-of-memory condition even for simple models, so I can't currently do more tests.

7eggert, 2017-05-20:
It was caused by the settings with 0.35 mm layer height. I sliced with 0.2 mm now:
[image]
[image]
[Unterteil.tar.gz](https://github.com/alexrj/Slic3r/files/1015835/Unterteil.tar.gz)

7eggert, 2017-05-20:
For the memory issue: Somehow the first layer height was switched from 0.3 mm to 0.3 % ...

dumle29, 2017-06-14:
I have a similar issue, and I think it fits in here.

With avoid crossing perimeters on, slic3r will generate toolpaths with a ton of retracts, crossing perimeters in this particular stl: https://mega.nz/#!6UMATQZQ!nGr02HGODP2yv8yP6Rtu0eANjTeMgvxtzfD8wj7NK9A

With avoid perimeters, the gcode looks like this:
[image]

Without it it's actually better:
[image]

Here's my Slic3r config bundle:
https://mega.nz/#!mINQGSQD!uZw_L9rGkWfhuVOcUFZxhuDL_xPspk01HHcdlQDP8tE

I'm using:

- BigDelta
- BigDelta - PLA
- BigDelta

Slic3r Version:
`Slic3r-master.2017.05.30.1143.6a7057b.64bit windows 10`

dumle29, 2017-06-14:
Got it. I'll edit to just export the relvant exports

foreachthing, 2018-04-20:
Still no improvement in Slic3r-master.2018.04.14.1655.7457cfd.64bit.
Sliced using Slic3r:
[image]

Slice with Cura
[image]

Yellow lines inside the hole: Travel.
The Actual part has all those tiny strings inside the hole. Not so much fun to clean that out.

File attached. Config: use _your_ config.
[Teil1_Standard.zip](https://github.com/slic3r/Slic3r/files/1930661/Teil1_Standard.zip)

ponzifex, 2019-11-13:
> Still no improvement in Slic3r-master.2018.04.14.1655.7457cfd.64bit.
> Sliced using Slic3r:
> [image]
>
> Slice with Cura
> [image]
>
> Yellow lines inside the hole: Travel.
> The Actual part has all those tiny [...]

What is this fancy looking slicer software?

foreachthing, 2019-11-13:
@ponzifex it's [CraftWare](https://craftbot.com/craftware/).
It does a good job with "support where I need it". It also slices really fast! It has some drawbacks as well ;-)
One thing, I used to use it for, was investigating gcode.