slic3r/Slic3r #3724: Slic3r plays a Russian Roulette on returning a Clipper::PolyTree object

bubnikv, 2017-02-22:
Slic3r plays a Russian Roulette on returning a Clipper::PolyTree object. Depending on how / whether the C++ compiler applies a return value optimization, a shallow copy of the PolyTree may be created, copying the pointers of child PolyNodes, then destruing the same object twice.

This happens on Visual Studio 2013 with optimization disabled to make C++ debugging feasible, where the current Slic3r master crashes on test cases.

One possible solution is to declare the move operators and disable the copy operators.

diff --git a/xs/src/clipper.hpp b/xs/src/clipper.hpp
index e60873b..de5568b 100755
--- a/xs/src/clipper.hpp
+++ b/xs/src/clipper.hpp
@@ -155,18 +155,35 @@ private:
     void AddChild(PolyNode& child);
     friend class Clipper; //to access Index
     friend class ClipperOffset; 
+    friend class PolyTree; //to implement the PolyTree::move operator
 };

 class PolyTree: public PolyNode
 { 
 public:
-    ~PolyTree(){ Clear(); };
+    PolyTree() {}
+    PolyTree(PolyTree &&src) { *this = std::move(src); }
+    virtual ~PolyTree(){Clear();};
+    PolyTree& operator=(PolyTree &&src) { 
+        AllNodes   = std::move(src.AllNodes);
+        Contour    = std::move(src.Contour);
+        Childs     = std::move(src.Childs);
+        Parent     = nullptr;
+        Index      = src.Index;
+        m_IsOpen   = src.m_IsOpen;
+        m_jointype = src.m_jointype;
+        m_endtype  = src.m_endtype;
+        for (size_t i = 0; i < Childs.size(); ++ i)
+          Childs[i]->Parent = this;
+        return *this; 
+    }

     PolyNode* GetFirst() const;
     void Clear();
     int Total() const;

 private:
-  //PolyTree& operator =(PolyTree& other);
-  PolyNodes AllNodes;
+    PolyTree(const PolyTree &src) = delete;
+    PolyTree& operator=(const PolyTree &src) = delete;
+    PolyNodes AllNodes;
     friend class Clipper; //to access AllNodes
 };

bubnikv, 2017-02-22:
[clipper.hpp.diff.zip](https://github.com/alexrj/Slic3r/files/792834/clipper.hpp.diff.zip)

alexrj, 2017-02-28:
It's not Slic3r, it's the Clipper library. Rather than patching it in the Slic3r codebase, why don't you report the issue upstream?