slic3r/Slic3r #3725: Infill/perimeters overlap is only overlapping in one axis

tcm0116, 2017-02-22:
### Version
1.2.9-628-g04d3f5b

### Operating system type + version
openSUSE 13.2

### Behavior
Infill/perimeters overlap is only overlapping in one axis.

The images below shows the infill around a circle with 0% overlap and then again with 100% overlap. As can be seen in the 100% overlap case, the infill overlaps at 100% when it is perpendicular to the perimeter. However, the overlap decreases to 0% as the infill becomes parallel to the perimeter.

[image]

[image]

Reviewing FillRectilinear.cpp, I suspect this issue is caused by the fact that the endpoints_overlap value is only added to the y component of the points in the polylines. Would it make more sense to offset the bounding polygons by the endpoints_overlap value on order to apply the offset in both the x and y directions?

bubnikv, 2017-02-22:
> Would it make more sense to offset the bounding polygons by the
endpoints_overlap value on order to apply the offset in both the x and y
directions?

This is how my initial FillRectilinear2 worked and you may test the result
with the Prusa3D builds.

tcm0116, 2017-02-22:
@bubnikv I was looking into adding the option for the infill/perimeter overlap for bridges per my thinking in issue 134 in your fork, and that's when I discovered this particular issue. Per our discussion, @alexrj has re-written the infill since then, so is your FillRectilinear2 compatible with the new infill code?

tcm0116, 2017-02-22:
It was actually easier to implement the change to offset the polygons that I thought it might have been.

https://github.com/tcm0116/Slic3r/commit/6b4b9f467f17ab6a59784234efeeb7f9794e349d

However, this change is causing fill.t to fail at line 300 (infill is only generated under the forced solid shells). I've been trying to understanding how this particular test works, but it's a little beyond my current understanding. Any ideas on why the test may be failing and if the test itself needs to be re-written for this change, or if I actually broke something?

lordofhyphens, 2017-02-22:
@tcm0116 just as a FYI:
A bugfix by @alexrj for infill overlap (may be relevant, espec
https://github.com/alexrj/Slic3r/commit/4a25b3e8befc119e7dad880ec4473a6ea460d28b

The redid rectilinear (based on rect2).
https://github.com/alexrj/Slic3r/commit/a881f755a1dc3a8b45a3a3a1f354297890fc3a08

You can see the test changes there from what they used to be. I hope that helps.

tcm0116, 2017-02-22:
@lordofhyphens Thanks for the info. This issue and my fix are with the current master, which includes the changes you've referenced. 

I looked at the test changes, but they don't appear to be in the area of the test that is failing with my change added.

lordofhyphens, 2017-02-22:
@tcm0116 Yeah, I included the commits so you could see what was different more plainly is all. 

If the test is failing and it wasn't changed with the alterations, then I'd probably look more closely at your changes. @alexrj is probably the person with the best working knowledge of the test. I haven't looked closely at it myself.