slic3r/Slic3r #3727: My Slic3r crash when I try to open it

paulorfmmb, 2017-02-23:
### Version
Latest stable version merge pull request #3713 

### Operating system type + version
Using Win 10
### Behavior
When I try to open it using the command
perl slic3r.pl
the program opens and crashes immediately.
using when building I get no errors.

alexrj, 2017-03-01:
Can you try the master branch?
The stable branch is a bit abandoned, and master is much more stable :-)

Tinchus2009, 2017-03-07:
Im running win 10 64b, I just builted the latest dev version: 1.2.9-677-gf4d73f1

It is working nice

alexrj, 2017-03-08:
@paulorfmmb, can you confirm the master branch works? Thank you

alexrj, 2017-03-13:
No answer from @paulorfmmb

closed 2017-03-13 by alexrj

paulorfmmb, 2017-03-29:
Sorry for not responding!!! I was able to make it run using perl 5.18 - 32bits
However I updated my slic3r to use perl 5.24 and now I'm having the same issue. The program closes after I try to open it.

lordofhyphens, 2017-03-29:
@paulorfmmb Everything has changed. 

Can you check the automatic build first? http://dl.slic3r.org/dev

Then check the wiki again and do a full setup again of perl and boost (as I said, much has changed).

Start slic3r from the command line. It will show you errors that way.

lordofhyphens, 2017-03-29:
@paulorfmmb FYI #3815

paulorfmmb, 2017-03-30:
I reinstalled everything and did everything from the beginning like in the wiki. Got same results.

lordofhyphens, 2017-03-30:
Do the downloads from dl.slic3r.org/dev work?

lordofhyphens, 2017-03-30:
Try to move your Slic3r directory to somewhere with no spaces in the path
or special characters. Ensure you have updated to the most recent version
with git pull.

Perl should be installed to C:\Strawberry. It also needs to be slic3r-perl.

I would also check command line slicing  (pass a path to the invocation).

paulorfmmb, 2017-03-30:
slic3r-perl is a directory?

lordofhyphens, 2017-03-30:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#slic3r-perl