slic3r/Slic3r #3728: Rectilinear top/bottom fill should be customisable in direction

PMunch, 2017-02-23:
This is a bit of a strange request. When using rectilinear infill it would be nice to be able to control the angle at which the lines were drawn. This primarily comes into play when printing objects which have a straight edge of some kind. Recently I tried to print a couple of hexagonal items for example, and the rectilinear lines didn't match up with any of the straight edges of the hex. This isn't that big of an issue but it would be nice to either have the option to change direction or to rotate the parts in other increments that 45 degrees.

alexrj, 2017-02-28:
There's a "Fill Angle" option. Did you try it?

PMunch, 2017-02-28:
So there is, consider this closed. And sorry for the bother.

closed 2017-02-28 by PMunch