slic3r/Slic3r #3729: Options for filament switching

PMunch, 2017-02-23:
The simplest way to achieve a multi-colour print on a single-head 3D printer is to cut the print at a specific layer, change filament, and keep printing. This is something which typically has to be done manually which is both tedious and error prone. It would be great if Slic3r had the ability to specify layer slices that would manifest itself as multiple .gcode files representing each filament. This would make multi-colour printing much easier. If this gets implemented it would also be great with an option for drawing the perimeter of a layer before stopping. This would allow e.g. [game tiles with a contrast colour layer](http://www.thingiverse.com/make:295377) to have the perimeter be the base colour to "hide" the layer change.

seanam, 2017-03-03:
Have you seen the [ColorPrint utility](http://www.prusaprinters.org/color-print/) from Prusa? I think this does what you want to do.

PMunch, 2017-03-03:
Hadn't seen it until now. But in their documentation they note things like the fact that they don't support z lifting etc. These are obvious limitations that come from reading the GCode generated by Slic3r without ant knowledge of what the different things in the file "means" verbose mode is helpful when doing manual cutting but Slic3r would be able to improve upon the multi-colour cutting capabilities because it has knowledge of how the file is cut, and the possibility to change the cutting based on colour.