slic3r/Slic3r #3730: Google Summer of Code 2017

alexrj, 2017-02-28:
I'm proud to announce Slic3r was accepted as a mentoring organization for GSoC 2017, thanks to an alliance with other open source projects such as BRL-CAD (leading the umbrella group), LibreCAD, OpenSCAD, STEPcode, LinuxCNC, FreeCAD. The philosophical purpose of this alliance is also to encourage and seek opportunities of working together on common code development.

GSoC is a global program that offers students stipends to write code for open source projects. They will learn coding and dealing with an open source projects, under our mentorship. It's a great opportunity for them but also for us as we might get some more work done and we might keep them as contributors also when the program ends.

Here's an overview of GSoC:
https://summerofcode.withgoogle.com/how-it-works/

Here's the CAx umbrella group page:
https://summerofcode.withgoogle.com/organizations/5733719949180928/

Here's the list of ideas that candidate students might want to apply for:
http://brlcad.org/wiki/Google_Summer_of_Code/Project_Ideas#Slic3r_Projects

If you want to suggest more ideas, do it here. We tried to identify tasks that don't need too heavy changes to the core parts, which might not be suitable for students.
If you want to help as a mentor (keep in mind this will take some weekly time) and you're familiar with the Slic3r codebase, you're more than welcome.

cenklar, 2017-02-28:
Hi, I was just looking where to post a suggestion for Slic3r and I came upon this, great news. First suggestion: to enable manual positioning of the object in the print bed; it´s useful when printing small parts and you want to avoid a certain region of the print bed (damaged, not clean enough, etc).

And second, it would be nice to be able to define a certain area/perimeter where the nozzle would never cross (kind of "avoid crossing perimeters 2.0". Of course, it´s even better if Slic3r can find the next-layer-start-point as close as the previous one, but sometimes "big jumps" are unavoidable. This feature would turn any oozing to point outwards the object´s surface, making the cleaning far more easy.

[image]

thanks for your work and time!

xoan, 2017-02-28:
@cenklar commented:

> First suggestion: to enable manual positioning of the object in the print bed; it´s useful when printing small parts and you want to avoid a certain region of the print bed (damaged, not clean enough, etc).

 File → Preferences → Auto-center parts (uncheck)

alexrj, 2017-02-28:
Yes, please do not use this for feature requests. Open distinct issues in the tracker. This is about GSoC.

lordofhyphens, 2017-03-30:
For students: The deadline for proposals is April 3.

https://developers.google.com/open-source/gsoc/timeline

@singh1114

lordofhyphens, 2017-03-30:
https://developers.google.com/open-source/gsoc/resources/manual#student_manual

lordofhyphens, 2017-05-04:
Congratulations to @Samir55 and @curieos on being accepted for GSoC!

https://summerofcode.withgoogle.com/projects/#6343944127381504
https://summerofcode.withgoogle.com/projects/#5662414690320384

@Samir55 will be adding 3MF support to Slic3r.
@curieos is tackling wireframe printing.

Good luck to both of you!

closed 2017-05-04 by lordofhyphens