slic3r/Slic3r #3733: Add support for custom gcode for triggering solenoid

seanam, 2017-03-03:
I'm working on a bioprinting project which uses a solenoid for extruding material. The process of modifying the g-code after slicing isn't very easy and am looking for ways to automate / improve it.

1. How difficult would it be to add "Starting extrusion" and "Stopping extrusion" sections to the Printer Settings -> Custom G-code area? This would be really useful for triggering the solenoid on / off.
2. Is there a way to prevent Slic3r from adding the E parameters to the g-code? I'm not using a motor so don't need this. Would it be possible to add a selection in the Filament settings to switch between motor extrusion and solenoid extrusion?
3. Another approach I thought of is using a post-processing script to delete the E parameters (not using a motor) and then and add the solenoid on/off code by check if movements (without E parameter) occurred before or after the code.

Any advice on how to implement this would be very appreciated. I don't have much experience with the slic3r code but can try to adding this if someone points me to the areas that need to be modified.

lordofhyphens, 2017-03-03:
#848 for a discussion of on/off. 

A "custom constant gcode extrusion" flavor would probably be appropriate for pumps/solenoids. It would need to enable a field to enter in custom on/off gcode that the _extrude() function would use to place the on/off. 

You wouldn't get the benefit of acceleration of course with what I have in mind.

lordofhyphens, 2017-03-03:
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches will have a copy of this implemented for testing purposes.

lordofhyphens, 2017-03-03:
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches should have a test implementation for this when it gets off of the build server, branch name `custom-constant-extrusion`

I've added a switch in the Extruders options, along with a couple gcode entry text boxes to use whatever command you want. 

Currently, the user is assumed to be able to match the flow rate from their weird extruder to the print speed.  Acceleration is not being set for print moves (although it is being set for travel afaik).

A possible refinement would be to override print speed with advanced knowledge of the flow rate instead. 

Please leave feedback in the PR thread.

lordofhyphens, 2017-03-03:
@seanam if you want to mess around with it check to see if the way I implemented it isn't good for that use case (too much pump action), there's an alternative which I can merge instead: `constant-extruder-fewer-pumps`

It will also be on the bintray link above.

seanam, 2017-03-03:
@lordofhyphens Thank you for your help! I will test both of these and leave feedback in the PR thread.

lordofhyphens, 2017-03-03:
@seanam if you look at the PR you can see what I modified. That + the wiki (for figuring out how to add new options) and you can try to do some more advanced work with it. I don't have that kind of printing setup here, so my ability to experiment is limited.

I suspect that overriding the print speed to suit a given flow would  be more useful.

lordofhyphens, 2017-03-03:
If you do more work on it you can open a pull request against the branch on *my* fork. I will look at it and add it to the main PR.

seanam, 2017-03-03:
@lordofhyphens I tried compiling this source from the custom-constant-extrusion branch (using instructions in How_to_build_Slic3r.txt file) but am getting an error. I've tried running different versions of boost (1.63, 1.60, 1.59) as mentioned in #692 (also tried with --force) but still get the ./xs error. Do you have any suggestions on how to solve this? I'm installing it on a mac (OSX 10.12.3)

Here is the log file contents:

cpanm (App::cpanminus) 1.7042 on perl 5.025010 built for darwin-thread-multi-2level
Work directory is /Users/sean/.cpanm/work/1488541015.38523
You have make /usr/bin/make
You have /usr/bin/curl
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
--> Working on ./xs
Entering /Users/sean/Slic3r/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Configuring /Users/sean/Slic3r/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Test::More 0 ... Yes (1.302078)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have Module::Build 0.3601 ... Yes (0.4220)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /Users/sean/.cpanm/work/1488541015.38523/build.log for details. Retry with --force to force install it.

I also tried to use the ones that you built from bintray but get the following error: 
perl slic3r.pl
Version mismatch between Carp 1.42 (/Users/sean/perl5/perlbrew/perls/perl-5.25.10/lib/5.25.10/Carp.pm) and Carp::Heavy 1.3301 (/Users/sean/Desktop/Slic3r/slic3r-constant-extruder-fewer-pumps/lib/Carp/Heavy.pm).  Did you alter @INC after Carp was loaded?
Compilation failed in require at local/lib.pm line 407.
BEGIN failed--compilation aborted at slic3r.pl line 9.

lordofhyphens, 2017-03-03:
The bintray.com build is for Windows. I don't have a build server up and
running yet for OSX. I will see what i can do about that this weekend.

Your issue is just one of the dependencies. I know @alexrj devs on OSX.

lordofhyphens, 2017-03-04:
@seanam https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X ?

seanam, 2017-03-07:
@lordofhyphens I wasn't able to fix the dependency issues on my mac but got it installed on a linux machine. I've tested it and it works well but it turns the solenoid on/off more frequently than necessary. The on/off commands are sandwiched before and after every G1 command that has extrusion instead of just at the beginning and end of continuous extrusion sequences. Here is some example output code to clarify:

;current Slic3r output for cube
G0 X40.000 Y40.000 Z0.200 ; move to starting point
M126                               ; ON
G1 X40.000 Y60.000       ; extrusion path 1
M127                               ; OFF
M126                               ; ON
G1 X60.000 Y60.000       ; extrusion path 2
M127                               ; OFF
M126                               ; ON
G1 X60.000 Y40.000       ; extrusion path 3
M127                               ; OFF
M126                               ; ON
G1 X40.000 Y40.000       ; extrusion path 4
M127                               ; OFF

;desired Slic3r output for cube
G1 X40.000 Y40.000 Z0.200 ; move to starting point
M126                               ; ON
G1 X40.000 Y60.000       ; extrusion path 1
G1 X60.000 Y60.000       ; extrusion path 2
G1 X60.000 Y40.000       ; extrusion path 3
G1 X40.000 Y40.000       ; extrusion path 4
M127                               ; OFF

If there was another print location (or part) on the same layer, I'd like it to turn off the solenoid, move to the location and then turn it on again like so:

;desired Slic3r output for cube with multiple extrusions for for same layer
G0 X40.000 Y40.000 Z0.200 ; move to starting point
M126                               ; ON
G1 X40.000 Y60.000       ; extrusion path 1, sequence 1
G1 X60.000 Y60.000       ; extrusion path 2
G1 X60.000 Y40.000       ; extrusion path 3
G1 X40.000 Y40.000       ; extrusion path 4
M127                               ; OFF
G1 X80.000 Y40.000       ; movement without extruding 
M126                               ; ON
G1 X80.000 Y60.000       ; extrusion path 5, sequence 2
G1 X100.000 Y60.000     ; extrusion path 6
G1 X100.000 Y40.000     ; extrusion path 7
G1 X80.000 Y40.000       ; extrusion path 8
M127                               ; OFF
; etc

Also, this is a minor issue but it still adds the G92 E0 commands even though I've turned off retraction in the settings (it inserts before each extrusion sequence). This shouldn't affect anything but it would be nice to turn it off if the "Custom Constant rate extruder" button is selected. Do you have an idea on what parts to change to fix these?

I've also attached the gcode output below.
[cube_solenoid_twoparts.txt](https://github.com/alexrj/Slic3r/files/824374/cube_solenoid_twoparts.txt)

seanam, 2017-03-07:
Sorry, i just realized maybe this is what `constant-extruder-fewer-pumps` does. Testing that now.

seanam, 2017-03-07:
I compiled the `constant-extruder-fewer-pumps` version and it builds fine but get an error when executing the program:
Error in option spec: "constant_extruder|constant-extruder!@" 
Do you know what this means?

lordofhyphens, 2017-03-07:
@seanam yeah, I screwed up the definition of a CLI option. It's fixed now (line 119 in xs/src/libslic3r/PrintConfig.cpp).

seanam, 2017-03-08:
I've gotten a chance to test both now, I'll leave comments in the pull request #3734 thread.