slic3r/Slic3r #3736: Feature request: avoid marring top infill surfaces with travel moves

DrLex0, 2017-03-04:
### Version
1.2.9

### Operating system type + version
Mac OS X 10.11.6

I would like to see either a checkbox _“avoid crossing top solid infill”,_ or _“Lift Z when crossing top solid infill”._ The reason can be seen in this photo:
[image]
Those diagonal lines are not supposed to be there. They are caused by the nozzle making a travel move across the surface right after printing it. The model is [this one](http://www.thingiverse.com/download:3338894) from [here](http://www.thingiverse.com/thing:1874675).  I obtained this smooth surface by enforcing a top solid infill width equal to my nozzle diameter (0.4mm), and printing it slowly (15mm/s) with the fan enabled.

I don't want to enable Z lift for an entire print just because it has a few of these surfaces that I want to come out perfectly. I don't really care how this is implemented, just that there is a way to avoid that the hot nozzle will plough through a freshly printed top surface. Z-lift which is only enabled when crossing such surfaces, seems to be the easiest approach.

bubnikv, 2017-03-04:
You also want to z-hop over overhangs, or try to avoid them.

DrLex0, 2017-03-04:
Good point, I have had many scary moments when the nozzle zapped at 140mm/s over a slightly curled-up overhang…

kefir-, 2017-03-04:
I think slic3r should try to avoid them if possible without moving farther than a threshold distance (or time-to-move, because of dripping), otherwise z-hop over. 

Perhaps the threshold time should be the time it takes to z-hop over, og you can move around the overhang faster than the z-hop, then move around. Otherwise do the z-hop. Also, if the calculation is very complex and takes too long, just z-hop.