slic3r/Slic3r #3739: Feature request: sequential printing without completing objects

Patola, 2017-03-05:
This is a feature present in Simplify3D, described here: https://www.simplify3d.com/support/articles/multi-part-printing/ (Multiple Process, Sequential Printing Mode)

### Version
I am currently using 1.3.0-dev 1.2.9-627-g7d2884b

### Operating system type + version
Ubuntu 16.04 64 bits

### Behavior
When using sequential printing, if objects are taller than the heigh clearance slic3r simply refuses to slice. But it should not be this way. It could behave like Simplify3D: since we are limited by this height, do a sequential printing for all objects up to a height of <height clearance>, then start over again at <height clearance+1> and do it sequentially again, then again at <height clearance x 2 + 1> and so on.

_Is this a new feature request?_
YES

Note: I did not include my configuration files because they aren't really needed, since this is a feature request. But I can of course provide them if deemed necessary.

[image]

fivethreeo, 2017-06-11:
Making a attempt at this, but my perl-fu is no good.

G-code generation starts but hangs for now.

https://github.com/fivethreeo/Slic3r/blob/master/lib/Slic3r/Print/GCode.pm#L227

lordofhyphens, 2017-06-11:
If you want to port it to c++ go nuts

fivethreeo, 2017-06-11:
->size->z is not units in mm it seems how to get mm units from it?

lordofhyphens, 2017-06-12:
There is a scaling factor function used to convert most distances.

lordofhyphens, 2017-06-14:
I don't have any objections to this fall-through behavior for sequential printing. @alexrj ?