slic3r/Slic3r #3740: Buttons Grey out incorrectly

thompsbp, 2017-03-07:
### Version
_Version of Slic3r used goes here_
1.3.0-Dev

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Unable to do this.

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 7 SP 1

### Behavior
* _Describe the problem_
Buttons grey out incorrectly.
Buttons are still useable and Export STL/GCODE buttons un grey out after use. Positioning buttons remain greyed out after use.

* _Steps needed to reproduce the problem_
Buttons are always greyed out after loading a new STL.

* _Expected Results_
Buttons will only be greyed out when not useable.

* _Actual Results_
Some Buttons are greyed out all the time. Others ungrey after use.

* _Screenshots from __*Slic3r*__ preview are preferred_
http://imgur.com/JRv2EeT

_Is this a new feature request?_
No

bubnikv, 2017-03-07:
I think this has something to do with the Windows styles, Windows
compatibility settings and the manifest of the application. My impression
is, it is a general wxwidgets issue. I can surely reproduce the issue on my
windows 7 computer. Slic3r behaves similarly on Windows 10, if the exe
wrapper misses a manifest, therefore the compatibility section is missing.

lordofhyphens, 2017-03-07:
Hm, the only button that I can reproduce the issue with on my (non-development) Win7 system with a recent build is the Arrange button.

alexrj, 2017-03-08:
I can't reproduce the issue, but the first fix I'd try would be to call `selection_changed()` (the method responsible for enabling/disabling buttons) within a `CallAfter()` callback. wxWidgets often has issues when changing GUI things within events triggered by wxWidgets itself. I'm not sure this applies to the C++ library as well or is specific to wxPerl.

gege2b, 2017-03-16:
my two cents.

I got the same issue on Windows 10 today (I used 1.2.9 until now)

But even if the buttons are greyed out, they can still be used (and if I remember, using one of them activate all the greyed buttons), at least on my PC

bubnikv, 2017-03-17:
On Windows 10 it works, if the application has a correct manifest. If you try to run the slic3r through the perl.exe from strawberry, it will be ok. If you do that through the slic3r.exe provided by @lordofhyphens , it may be wrong as it may not contain the necessary manifest. If you run the Slic3r Prusa Edition, it should be fine.

lordofhyphens, 2017-03-17:
Lower: run by wrapper script. upper: run script through perl5.18.4.exe (same dlls)
[image]
Yeah, it's something with how the wrapper exe works (I note that it's as what's provided in the prusa3d fork).

bubnikv, 2017-03-17:
You may use the resource hacker to analyze and replace the manifest.
http://www.angusj.com/resourcehacker/

lordofhyphens, 2017-03-17:
I'll have to examine what is going on (#thankswindows) and bake whatever changes that need to be made into the build process.

closed 2017-03-18 by lordofhyphens

bubnikv, 2017-03-18:
> Closed #3740 <https://github.com/alexrj/Slic3r/issues/3740> via d4136b3

This solves the behavior on Windows 10, not in Windows 7 IMO.

lordofhyphens, 2017-03-18:
[image]
That's from a Windows 7 system, which is the expected look.

commit d4136b38 (2017-03-18): Add manifest to Windows RC file (adapted from Strawberry Perl manifest).
Fixes #3740