slic3r/Slic3r #3741: Small perimeter speed setting being ignored/over written by a faster 1st layer speed?

Bluelinegecko, 2017-03-07:
### Version

Version 1.2.9

### Operating system type + version

Windows 10
### Behavior

* _Describe the problem_
When setting printing speeds manually, and using a definitive numerical value, slic3r seems to ignore speeds lower than the speed entered in the "first layer speed" modifier. Small perimeters print at a much higher speed (30mm/s or whatever is entered) than the default (15mm/s or whatever is entered) small perimeter speed only on the first layer. Small perimeters then print at the correct set speed during the remaining layers. 

* _Steps needed to reproduce the problem_

Slice a file with the first perimeter speed set higher than the speed set for small perimeters. 
 
* _Expected Results_

One would expect slic3r to select the slowest of the 2 speed settings when printing small perimeters on the first layer as a slower speed is both likely to reproduce small perimeters more accurately, and more importantly, improve the adhesion of small details to the print bed. Printing small perimeters at 30mm/s as in the default first layer speeds is more likely to cause those perimeters to lift from the printbed before the layer is completed. This might also cause problems with printing the first layer of support material if the pattern is set to pillars or it has a footprint <=6.5mm used to trigger the small perimeter speed.

* _Actual Results_

Slic3r ignores the small perimeter speed setting for the 1st layer and prints the entire layer at the same speed. I have not confirmed whether or not this occurs if the 1st layer speed is set to a percentage value of the other layer speeds rather than a definitive speed setting.

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

No. 

Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Tinchus2009, 2017-03-07:
I think this is how it is supose to work. Slic3r will ignore all speed setting for the first layer. In the fist layer the speed will be the one set as "first layer speed", wich in general is suposed to be very sow in order to get a good first layer

Bluelinegecko, 2017-03-07:
Tinchus,

I understand that. And for every other speed parameter you can adjust the "first layer" setting is going to be slower. But with very small perimeters, especially circular ones where the extrusion tends to be pulled inward towards the center of the circle, printing at even slower speeds that one would normally print the first layer is recommended for better accuracy. I believe that is the reason that Slic3r's default "small perimeter" speed is 1/2 of the default "First layer" speed. It really makes no sense to watch a printer fly through printing a bunch of small perimeters on the 1st layer at 30mm/s, and then slow down and print them at a slow 15mm/s for the rest of the model. Why not have the code look at any manually set speeds and use the first layer speed unless a particular parameter is slower?

alexrj, 2017-03-08:
@Bluelinegecko, your proposal about turning First Layer Speed into Max First Layer Speed makes sense.

spit40, 2019-03-26:
Exactly the problem I'm having

supermerill, 2019-03-27:
you can merge that 5d92ab481f0936b7c0ca2637b7793091596ce04b